toblum/McLighting #198: Coudnt finde "/stripstate.json"  Fehler!

Cifi76, 2018-07-05:
Hallo, auch Ich habe das gleiche Problem

Coudnt finde "/stripstate.json" 
Fehler!

Die LEDs reagieren erst gar nicht. 
Ich habe es wie folgt geschlossen

VIN = LED 5V 
G = LED - 
D5 = Steuerkanal

Im Skript sieht das so aus:

// Neopixel 
#define PIN D5 // PIN (14 / D5) wo der neopixel / WS2811 Strip angeschlossen ist 
#define NUMLEDS 10 // Anzahl der LEDs im Strip 
#define BUILTIN_LED 2 // ESP-12F hat die eingebaute LED auf GPIO2 , siehe esp8266 / Arduino # 2192
#define BUTTON 4 // Eingangspin (4 / D2) zum Ein- / Ausschalten des LED-Streifens, verbinden Sie diesen PIN mit Erde, um den Knopf auszulösen.

Warum brennen die LEDs nicht? 
Wo liegt der Fehler?

Danke für deine Hilfe

toblum, 2018-07-05:
Hallo @Cifi76,

die Meldung mit "stripdate" ist kein Fehler in dem Sinn, das tritt nur am Anfang auf, wenn der Modus noch nicht gewechselt wurde. Das kannst du in der Regel ignorieren.

Von dem was du schreibst, sollte der Strip korrekt angeschlossen sein. Die genannten Einstellungen stimmen.

Folgende Fragen:
- Konntest du die WLAN Zugangsdaten eintragen?
- Hast du weitere Änderungen am Skript vorgenommen?
- Sende bitte auch einmal die komplette Ausgabe des seriellen Monitors.
- Die LEDStrips sind recht empfindlich. Du solltest sie zudem nicht im Betrieb ein/ausstecken. Dabei könnten sie Schaden nehmen.

Viele Grüße
Tobias

Cifi76, 2018-07-05:
Hallo, und Danke für die Rückantwort. Ich konnte die Zugangsdaten einrichten und komme nun über die IP http://192.168.178.54/ auf die Web Oberfläche. Wenn Ich hier etwas ändere sehe Ich das am seriellen Monitor. Ich habe es gerade noch einmal mit neuen LEDs versucht. Liegt es vielleicht an den LEDs Ich verwende die WS2812B ? 
 
Hier ein kleiner Auszug:
handleFileRead: /
WS: [0] Disconnected!
WS: [0] Connected from 192.168.178.31 url: /
WS: [0] get Text: *00E1FF
WS: Set all leds to main color: [0] [225] [255]
WS: Set main color and light all LEDs [*00E1FF]
SET_MODE: 0 0
Saving cfg: {"mode":1,"strip_mode":0,"brightness":57,"speed":196,"red":0,"green":225,"blue":255} Success!
MQTT: Send [home/Fahrrad-Schmiede_ha/state/out]: {"state":"ON","color":{"r":0,"g":225,"b":255},"brightness":57,"color_temp":327,"speed":196,"effect":"Static"}
Heap size: 22192
WS: [0] get Text: *FF8000
WS: Set all leds to main color: [255] [128] [0]
WS: Set main color and light all LEDs [*FF8000]
SET_MODE: 0 0
Saving cfg: {"mode":1,"strip_mode":0,"brightness":57,"speed":196,"red":255,"green":128,"blue":0} Success!
MQTT: Send [home/Fahrrad-Schmiede_ha/state/out]: {"state":"ON","color":{"r":255,"g":128,"b":0},"brightness":57,"color_temp":327,"speed":196,"effect":"Static"}
Heap size: 22360

Was mir gerade aufgefallen ist, wenn Ich das ganze ans Fahrrad bauen möchte ..... Über mein Handy kann Ich mich ja gar nicht verbinden ? Da kann Ich die IP nicht aufrufen. Kann man da statt der IP auch einen Namen einprogrammieren ? Denn sonst kann Ich das ganze ja nur über den PC steuern.

Das Projekt ist echt toll, und es ist echt der Hammer das es solche Personen gibt die so etwas auf die Beine stellen. Danke

Cifi76, 2018-07-06:
Habe wieder ein wenig weiter versucht, und habe festgestellt das an der LED nur 1,2 Volt ankommen ? Ich habe das ganze über einen USB Port am PC angeschlossen ist das so richtig ? Sorry aber Ich bin ein wirklicher Anfänger aber bin begeistert davon.

Danke Gruß Michael

Cifi76, 2018-07-06:
Habe jetzt einfach mal die WS2812W direkt an eine 5Volt Spannung angeschlossen und siehe da es blinkt und leuchtet :-) Der Strom vom USB Port reicht wohl nicht aus um die LEDs anzusteuern. Soweit ist mir nun alles klar. Wenn Ich nun die Sache ans Fahrrad anschließe habe Ich das Problem das Ich über die IP nicht auf die Nutzeroberfläche komme. Wie kann Ich nun mein Handy mit dem System koppeln ?

Güße Michael

toblum, 2018-07-06:
Hallo! @Cifi76,

schön, dass es nun klappt. Die Spannungsversorgung muss natürlich passen. Wieviele LEDs hast du und wo genau hattest du sie bisher angeschlossen? Wenn es nur wenige LEDs sind, dann kann man sie meist auch direkt mit den NodeMCU Boards versorgen.

McLighting braucht aktuell ein WLAN-Netzwerk, um zu funktionieren. Unterwegs kannst du aber mit deinem Smartphone ein WLAN aufmachen und McLighting damit verbinden. Das funktioniert auch gut.

In der Regel kannst du McLighting auch immer über den Hostnamen statt IP aufrufen. Wenn als Hostname "McLighting01" eingestellt ist, dann solltest du McLighting auch über http://McLighting01.local erreichen können.

Viele Grüße
Tobias

Cifi76, 2018-07-06:
Hallo Tobia, danke für die Rückmeldung. Wenn Ich aber unterwegs bin wie soll Ich mich da mit einem Netzwerk verbinden ? Gibt es so etwas wie ein Netzwerkdummy ? Es waren 10 x 5V LEDs angeschlossen jedoch kamen nur 1,8 V am VIN Anschluss raus. Denke mal das der USB vom PC nicht genug Leistung hatte. 

Ansonsten echt eine super Sache die so langsam richtig Spaß macht.

Grüße Michael

toblum, 2018-07-06:
Hallo!

Wie gesagt: Eigentlich alle Smartphones können ein eigenes WLAN-Netzwerk aufmachen. Stichwort "Mobiler Hotspot" / "Tethering". Damit hast du dein Netzwerk quasi dabei und verbindest dann McLighting mit diesem Netzwerk statt mit deinem Heimnetzwerk.

10 LEDs sollten noch locker direkt über das NodeMCU-Board funktionieren. PC-USB Anschlüsse können aber tatsächlich zu schwach sein. Wenn die Software jetzt ja läuft, dann kannst du es ja mal mit einem anderen USB-Netzteil testen.

Viele Grüße

Cifi76, 2018-07-06:
Hallo, soweit verstehe Ich die Logik, aber wie kann Ich dann die index.html hochladen ? Das verstehe Ich noch nichts so ganz,

Gruß

emtantra, 2018-07-08:
if this discussion on English then it could be useful to all of us.

toblum, 2018-07-21:
Hallo Michael,

wenn dein normales WLAN nicht erreichbar ist, dann geht McLighting wieder in den AccessPoint Modus (LEDs leuchten blau). Dann verbindest du dich mit dem McLighting WLAN und rufst die 192.168.4.1 auf. Dort gibst du diesmal die Zugangsdaten deines Handy-Hotspots ein. Zukünfig verbindet sich McLighting immer mit deinem Handy-Hotspot und du kannst ihn darüber erreichen und steuern.

Viele Grüße

toblum, 2018-12-16:
Closing issue due to inactivity.

closed 2018-12-16 by toblum