toblum/McLighting #199: invalid initialization of non-const reference of type

Fillip38, 2018-07-05:
Hallo,
beim Überprüfen habe ich eben folgenden fehler erhalten:

invalid initialization of non-const reference of type 'ArduinoJson::JsonObject&' from an rvalue of type 'ArduinoJson::Internals::enable_if<true, ArduinoJson::JsonObject>::type {aka ArduinoJson::JsonObject}'

Update auf JSON 6.1.0-beta habe ich bereits durchgeführt, wo könnte der Fehler liegen?
[invalid initialization of non-const reference of type.txt](https://github.com/toblum/McLighting/files/2167309/invalid.initialization.of.non-const.reference.of.type.txt)

Fillip38, 2018-07-05:
Ich dachte die Installation / Aktualisierung von esp8266 hätte den Erfolg gebracht, leider besteht das Problem immer noch

debsahu, 2018-07-05:
I dont speak german, but from what I could translate I see you are having issues compiling. There were [breaking changes in ArduinoJson 6.1.0-beta](https://github.com/bblanchon/ArduinoJson/releases) and the relevant code changes are reflected in [McLighting develop branch](https://github.com/toblum/McLighting/tree/develop). Regarding the errors with ticker, upgrade esp8266 core to latest v2.4.1.

Fillip38, 2018-07-06:
Ich habe ab Version 6 nur die Beta Versionen als Möglichkeit zur Installation...