toblum/McLighting #200: Weird color outputs using NeoAnimationFX

Nobodi99, 2018-07-08:
Hello :)

So far I have to say it'sa really cool project and I like the idea of it.
I recently built some LED stuff for my home and I was using the WS2812FX library so far to achieve some light effects.
Last week I stumbled over this project and was thrilled to try it out. Especially because it implements a library to control the animations with the NeoPixelBus library. 
After some trial and error and some starting problems I have everything up and running but still there's one more error left.

When I output the effects to the LEDs I get some weird light effects which are nowhere near the desired outcome.

Here is a picture of how the leds look like when I try to display a simple static red color.
And it keeps changing some of the leds at random.

[image]

I do know that the build itself works because I already tested it with the NeoPixelBus library and a small test program. But as soon as I use this project it keeps giving me this outcome.

Would really appreciate some tips to fix this :)

Keep up the good work and greeting,
Daniel

debsahu, 2018-07-08:
What's your LED type? Share your NeoPixelBus code? But random colors are due to noisy power supply, adding a 1000uF capacitor helps. What pin are you using RX?

Nobodi99, 2018-07-08:
The LEDs I use are some SK6812 LEDs I ordered from china. But the are more or less WS2812 with just GRB.

I used this code just to test if the desired output will be achieved.
I'm using the RX Pinfor output.

`
#include <NeoPixelBus.h>

const uint16_t PixelCount = 30;

#define colorSaturation 128

NeoPixelBus<NeoGrbFeature, Neo800KbpsMethod> strip(PixelCount);

RgbColor red(colorSaturation, 0, 0);
RgbColor green(0, colorSaturation, 0);
RgbColor blue(0, 0, colorSaturation);
RgbColor white(colorSaturation);
RgbColor black(0);

void setup()
{
    Serial.begin(115200);
    while (!Serial);
    
    strip.Begin();
    strip.Show();
}

void loop()
{
    delay(500);

    for (int i = 0; i < 30;)
    {
      strip.SetPixelColor(i, red);
      strip.SetPixelColor(i+1, green);
      strip.SetPixelColor(i+2, blue);
      strip.SetPixelColor(i+3, white);
      strip.SetPixelColor(i+4, black);
      strip.Show();

      i = i+5;
    }

    delay(500);

    for (int i = 0; i < 30; i++)
    {
      strip.SetPixelColor(i, black);
      strip.Show();
    }

    delay(500);
}
`

debsahu, 2018-07-08:
That's excellent input. That code is pretty much what is being used in NeoAnimationFX. I'm stumped as to why that code works and not McLighting!

Hardware wise, are the strips and esp8266 on common ground? I also find that 3.3V signal may not trigger some chips, do you use a 3.3-5V level shifter?

Nobodi99, 2018-07-08:
The strips are connected to a nodemcu V2 board and both are powered by the same 5V/4A power supply. 
I ran the code I posed on the exact same setup which I used to take the picture I posted.
That's why I don't think it's a hardware issue. But I may be wrong.

I'm pretty new to Arduino/C++ programming so I'm a bit clueless here.
I will try to break down the code a bit more and delete everything I don't need/use and try to pinpoint the reason this happens.

I may also build a small test circuit and try to use the NeoAninationFX library without the McLighting packaging just to see where the error occurs.

I just hope it will be fixable.

debsahu, 2018-07-08:
Keep me in the loop

debsahu, 2018-07-08:
Could you try this:

```
#include <NeoAnimationFX.h>

#define NEOMETHOD NeoPBBGRB800
#define numPixels 30

NEOMETHOD strip(numPixels);
NeoAnimationFX<NEOMETHOD> myPixelRef(strip);

void setup() {
  myPixelRef.init();
  myPixelRef.setColor(255,0,0);
  myPixelRef.setMode(FX_MODE_STATIC);
  myPixelRef.start();
}

void loop() {
  myPixelRef.service();
}
```

Nobodi99, 2018-07-09:
The code works flawlessly and displays a nice red. 

I compiled the code at work today with a smaller strip (4 LEDs) of the same 5m reel I cut the ones I used here from. It worked like a charm. Next step will be to remove all libraries and install them again. Will see how that turns out. Seems to be a problem with my setup at home somehow. It's just weird that the colors in both test codes work but the MCLighting project doesn't...

Nobodi99, 2018-07-09:
Alright....It seems like there occured a problem when I updated the ArduinoJson library and something got screwed up rather badly which produced this error.

I deleted the library and installed it again and now it seems to work fine. 

Thank you for your help and patience :)

Nobodi99, 2018-07-09:
I lied.... just switched back to run the circuit with a dedicated power supply and error is back...

When I use USB it works and I already added a capacitor parallel to the supply. I really don't know the reason...

debsahu, 2018-07-09:
Ha then this issue is with your power supply! Funky colors means the power is still noisy, try adding a few capacitors of different values and types. Nothing to do with this code or NeoAnimationFX or NeoPixelBus, right?

debsahu, 2018-07-09:
Also could you provide a schematic of your circuit? Maybe pictures?

Nobodi99, 2018-07-09:
Yes seems like the code is fine. 
I'm just iritated because I use the same setup but only using the WS2812FX library and it works without issues. Hmm....

I attached a picture of my circuit. It's handdrawn so it's a bit messy. But it should be readable I hope :D 

The wiring is rather messy to look at but I tested it and it is fine. No short circuits or anything fishy there...

I mean I do work with electronics a lot but I seem to fail here.

[image]
[image]

debsahu, 2018-07-09:
Again fantastic pictures. Your circuit looks fine to me. But you mentioned when running on USB, the strip works and on power supply you get funky colors. That's leading me to think the power supply is not giving 5V, could you check that for me while LEDs are on.

Also on software side, in definitions.h
```
//#define USE_NEOANIMATIONFX
#define USE_WS2812FX
#define PIN 2 //GPIO2/RX pin
```

Nobodi99, 2018-07-09:
I just measured the voltage and I do get 5V on the LEDs.

When I change to the WS2812 library it doesn't do anything... The LEDs stay black.

debsahu, 2018-07-09:
I have to ask this, do you another power supply?

Nobodi99, 2018-07-09:
Yes I have and I tested it...
I have a 5V/10A  switching power supply and when I use that it works fine....

WHich I still don't understand because I have another project here and it uses the WS2812 library and the same power supply (I already exchanged them to see if that works) and it works fine...
That's why I'm frustrated right now :/

But definitely thanks for the help :)

Do you have an idea if I could stabilize the power supply somehow?

debsahu, 2018-07-10:
Sorry that you are facing such issues, power supplies are flaky especially the ones from far east. I have cycles through 3 PSUs for my desktop through a reputable supplier/maker of PSU. I have a boost converter that converts power from 18650 for a neopixel, and that is extremely noisy leading to random colors. What I did was added 10uF ceramic, 100uF electrolytic and 1000uF electrolytic capacitors in parallel and the noise is almost gone. Sometimes I do see a flicker but mostly stable.

Since there is no issue with the code, closing.

closed 2018-07-10 by debsahu