toblum/McLighting #201: 'DynamicJsonDocument' was not declared in this scope

fm14beck, 2018-07-09:
help me, why after doing the compiler, sketch to issue error. this is my first time trying to learn arduino, please give me enlightenment

debsahu, 2018-07-09:
Please install ArduinoJSON 6.0.1-beta

fm14beck, 2018-07-09:
after install ArduinoJSON i get " invalid initialization of non-const reference of type 'ArduinoJson::JsonObject&' from an rvalue of type 'ArduinoJson::Internals::enable_if<true, ArduinoJson::JsonObject>::type {aka ArduinoJson::JsonObject}'

what happend

debsahu, 2018-07-09:
I think you installed ArduinoJSON 6.1.0-beta instead! There are huge breaking change between 6.0.1 and 6.1.0.

debsahu, 2018-07-10:
As of now, the latest McLighting requires latest ArduinoJSON (6.1.0-beta). Keeping up with the breaking changes is a pain. Closing.

closed 2018-07-10 by debsahu