toblum/McLighting #206: Error when enabling AMQTT and Home assistant support

aidbish, 2018-07-10:
Hi 
When enabling the following items 
#define HTTP_OTA             .
#define ENABLE_AMQTT     
//#define ENABLE_MQTT      
#define ENABLE_HOMEASSISTANT 
#define ENABLE_BUTTON      
#define MQTT_HOME_ASSISTANT_SUPPORT 

i get the following error.

In file included from C:\Users\adrian\Desktop\mclighting\McLighting-master\Arduino\McLighting\McLighting.ino:218:0:

sketch\request_handlers.h: In function 'void onMqttConnect(bool)':
request_handlers.h:1036: error: 'class ArduinoJson::JsonObject' has no member named 'measureJson'
           char buffer[json.measureJson() + 1];
                            ^
request_handlers.h:1037: error: 'buffer' was not declared in this scope
           serializeJson(json, buffer, sizeof(buffer));                     
exit status 1
'class ArduinoJson::JsonObject' has no member named 'measureJson'

i have tried different versions of arduiniojson but the error persists

Any ideas

fm14beck, 2018-07-10:
Please install ArduinoJSON 6.0.1-beta

aidbish, 2018-07-10:
Already have that installed, tried a few different versions, the erro ronly appears when enabling
#define MQTT_HOME_ASSISTANT_SUPPORT

debsahu, 2018-07-10:
Thanks for pointing that out, I'll fix this, measureJson was changed in ArduinoJSON 6.x.x

debsahu, 2018-07-10:
@aidbish I have pushed the fix, but all that is change `json.measureJson()` to `measureJson(json)`

closed 2018-07-10 by debsahu

debsahu, 2018-07-10:
Thanks, now the [Wiki link](https://github.com/toblum/McLighting/wiki/Software-installation) should reflect this requirement. Yesterday's McLighting merge brought the requirement up to the latest ArduinoJSON release.