toblum/McLighting #210: group leds

psander247, 2018-07-21:
Hey
i would like to build a chain of several   lightfixures, made of WS2812b led strips. 5 pixels per lamp. 
Is it possible to group each 5 pixels so that they show up in the same color? 
That would be nice for rainbow effects and chasing lights as well. 

TIA 
Greetings from Germany
Patrick

toblum, 2018-07-21:
Hi @psander247,

I'm not completely sure what you're planning to do. But did you see Brians video here: https://youtu.be/T1Ul7iN72lI?t=36 He did something that sounds similar to me.

Regards
Tobias

psander247, 2018-07-21:
Hey @toblum 
that was fast :+1: 

Its kind of like Brians solution. 

The difference is, that I´m planning to link them with the data-line and controll them with one esp (Like if it was just one stripe of leds). Each of the lamps will contain 5 leds of the stripe. So it would be cool, if they would have the same color in the animation. 

Hope that is understandable.

psander247, 2018-07-21:
@toblum 
Thanks for the fast reply. 
Than i will try my best doing some changes to the effect library. 

Keep going, very good project, thanks for sharing :)

toblum, 2018-07-22:
Hi @psander247,

I forgot something: Maybe the range commands in the web socket API may help you:
https://github.com/toblum/McLighting/wiki/WebSocket-API

Regards 
Tobias