toblum/McLighting #211: Setting NUMLEDS as a parameter

mpugazh, 2018-07-22:
Hi Team,

Is it possible to set the NUMLEDS constant as a parameter so as to enable us to send number of led's over a post request? Also is it possible to send some value and parse that value inside the mclighting.ino.

For example, I would like to send a light intensity to a variable and convert that light intensity corresponding to number of led's(NUMLEDS)

Thanks and regards,

Pugazhendhi M

toblum, 2018-07-22:
Hi

I’m not completely sure what you’re trying to achieve. To light only a certain amount of leds, you may use the range commands over web sockets: https://github.com/toblum/McLighting/wiki/WebSocket-API

Regards 
Tobias

mpugazh, 2018-07-23:
I think i got it now. Closing this,

closed 2018-07-23 by mpugazh