toblum/McLighting #212: Trouble compiling

dinki, 2018-07-23:
I _think_ I have all libraries installed correctly and I'm getting compile errors I don't know how to overcome:

`Arduino: 1.8.5 (Windows 7), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"

In file included from C:\Users\User\Desktop\McLighting\McLighting.ino:218:0:

sketch\request_handlers.h: In function 'void checkpayload(uint8_t*, bool, uint8_t)':

request_handlers.h:414: error: 'class Ticker' has no member named 'active'

     if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                      ^

request_handlers.h:417: error: 'class Ticker' has no member named 'active'

       if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                             ^

request_handlers.h:434: error: 'class Ticker' has no member named 'active'

       if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                        ^

request_handlers.h:464: error: 'class Ticker' has no member named 'active'

       if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                        ^

request_handlers.h:467: error: 'class Ticker' has no member named 'active'

       if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                             ^

request_handlers.h:489: error: 'class Ticker' has no member named 'active'

       if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                        ^

request_handlers.h:492: error: 'class Ticker' has no member named 'active'

       if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                             ^

request_handlers.h:572: error: 'class Ticker' has no member named 'active'

         if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                          ^

request_handlers.h:575: error: 'class Ticker' has no member named 'active'

         if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                               ^

request_handlers.h:645: error: 'class Ticker' has no member named 'active'

       if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                        ^

request_handlers.h:648: error: 'class Ticker' has no member named 'active'

       if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                             ^

sketch\request_handlers.h: In function 'void onMqttMessage(char*, char*, AsyncMqttClientMessageProperties, size_t, size_t, size_t)':

request_handlers.h:900: error: 'class Ticker' has no member named 'active'

         if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                          ^

request_handlers.h:902: error: 'class Ticker' has no member named 'active'

           if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                                 ^

sketch\request_handlers.h: In function 'void shortKeyPress()':

request_handlers.h:1088: error: 'class Ticker' has no member named 'active'

         if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                          ^

request_handlers.h:1091: error: 'class Ticker' has no member named 'active'

         if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                               ^

request_handlers.h:1104: error: 'class Ticker' has no member named 'active'

         if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                          ^

request_handlers.h:1107: error: 'class Ticker' has no member named 'active'

         if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                               ^

sketch\request_handlers.h: In function 'void mediumKeyPress()':

request_handlers.h:1124: error: 'class Ticker' has no member named 'active'

       if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                        ^

request_handlers.h:1127: error: 'class Ticker' has no member named 'active'

       if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                             ^

sketch\request_handlers.h: In function 'void longKeyPress()':

request_handlers.h:1143: error: 'class Ticker' has no member named 'active'

      if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                       ^

request_handlers.h:1146: error: 'class Ticker' has no member named 'active'

       if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                             ^

C:\Users\User\Desktop\McLighting\McLighting.ino: In lambda function:

McLighting:611: error: 'class Ticker' has no member named 'active'

       if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                        ^

McLighting:614: error: 'class Ticker' has no member named 'active'

       if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                             ^

C:\Users\User\Desktop\McLighting\McLighting.ino: In lambda function:

McLighting:639: error: 'class Ticker' has no member named 'active'

         if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                          ^

C:\Users\User\Desktop\McLighting\McLighting.ino: In lambda function:

McLighting:690: error: 'class Ticker' has no member named 'active'

       if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                             ^

C:\Users\User\Desktop\McLighting\McLighting.ino: In lambda function:

McLighting:713: error: 'class Ticker' has no member named 'active'

       if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                             ^

C:\Users\User\Desktop\McLighting\McLighting.ino: In lambda function:

McLighting:733: error: 'class Ticker' has no member named 'active'

         if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                               ^

C:\Users\User\Desktop\McLighting\McLighting.ino: In lambda function:

McLighting:752: error: 'class Ticker' has no member named 'active'

         if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                               ^

C:\Users\User\Desktop\McLighting\McLighting.ino: In lambda function:

McLighting:771: error: 'class Ticker' has no member named 'active'

         if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                               ^

C:\Users\User\Desktop\McLighting\McLighting.ino: In lambda function:

McLighting:790: error: 'class Ticker' has no member named 'active'

         if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                               ^

C:\Users\User\Desktop\McLighting\McLighting.ino: In lambda function:

McLighting:809: error: 'class Ticker' has no member named 'active'

         if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                               ^

C:\Users\User\Desktop\McLighting\McLighting.ino: In lambda function:

McLighting:828: error: 'class Ticker' has no member named 'active'

         if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                               ^

C:\Users\User\Desktop\McLighting\McLighting.ino: In lambda function:

McLighting:847: error: 'class Ticker' has no member named 'active'

         if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                               ^

C:\Users\User\Desktop\McLighting\McLighting.ino: In lambda function:

McLighting:869: error: 'class Ticker' has no member named 'active'

       if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                        ^

McLighting:872: error: 'class Ticker' has no member named 'active'

       if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                             ^

exit status 1
'class Ticker' has no member named 'active'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

This is a screenshot of my installed libraries:

https://imgur.com/a/fWow9Ha

Can someone tell me what might be doing wrong?

debsahu, 2018-07-23:
Update your arduino-core (esp8266-arduino) to version 2.4.0 or higher. `ticker.active()` was introduced only recently.

dinki, 2018-07-23:
Thanks for the quick reply.  I don't see that when I search for arduino-core or esp8266-arduino in my library manager.  I appreciate your patience but can you tell me what I need to do to update this?

debsahu, 2018-07-23:
please follow instructions in https://github.com/toblum/McLighting/wiki/Software-installation

dinki, 2018-07-23:
I got it to build.  Thank you for your patience.  I guess I was focused too much on the libraries that I missed the important stuff on the top of the page.  I'm excited to give this a try!

closed 2018-07-23 by dinki