toblum/McLighting #214: Turn off command from app doesnt work correctly.

dimitris17563, 2018-07-30:
Hi,
First i want to thank you for the very nice project you share with us. I wanted to make a wi-fi E27 bulb controling it from mobile so your project is ideal for it. I managed everything to work fine althought i had some problem with not stable operation of leds (some flickering) which was because of bad connection with signal and also i reduced the resistor to 100ohms and now works fine. I have only one problem. When i click the OFF button in app the leds turn off but after 1-2 seconds turn on again. How can i fix it. I have not install the button because i will not have a contact with bulb, i want everything to be remotely.
Thank you very much for your help.

Best Regards
Dimitris

toblum, 2018-08-14:
Hi @dimitris17563,

I must have missed you issue. At first I don't have any idea. I think I never saw this happen before.
For debugging it would be nice to have the serial console output when that happens. Do you think you can have a look at it?

Regards
Tobias

dimitris17563, 2018-08-15:
Hi Tobias,
Thank you for your answer. Do you mean that i must have connected on computer and watch the serial monitor while i use the device? 
Best regardsDimitris

 
 

  
 
|  | Απαλλαγμένο από ιούς. www.avast.com  |

toblum, 2018-08-16:
Yes, that would hep to see what happens. But you have to find a way to power the ESP not from mains to connect your computer safely.

dimitris17563, 2018-08-16:
I have a usb dongle with 3.3V out put and i will connect to computer as i did for proggraming. I have not yet completed the project and its without mains power yet, because i was waiting to solve that problem with turn off.When i will have news i will send you message. Thank you.

dimitris17563, 2018-08-18:
Hi Tobias,
So i have the port monitor screen shot. When i click on the turn off button in app i saw the line which starts "Saving cfg:.........." and lights turn off for about 1 sec and then turn on and continue is same status.
What do you think is wrong?
Thank you.

[image]

dimitris17563, 2018-08-18:
Hi Tobias,
I have the screen shot of the port monitor attached below. When i click the turn off button you i saw the line which starts with "Saving conf : ..........." and lights turn off for about 1 sec and then turn on again and continue in the same mode. What do you think is wrong?
Thank you for your time.

[image]

toblum, 2018-08-19:
Hi @dimitris17563,

this looks OK as far as I can see. When you change a mode, it's saved after 1 second so that it keeps the mode when you restart the ESP. But saving should nnot reset something. I think it has something to do with it.

- What are your settings in definitions.h for ENABLE_STATE_SAVE_SPIFFS and ENABLE_STATE_SAVE_EEPROM.
- When you set a mode and restart the ESP, does it continue with that mode or do you see the rainbow animation again?
- What is the output of: "http://<your ESP IP>/esp_status"?
- What are your settings for the board in the Arduino IDE?

Regards
Tobias

dimitris17563, 2018-08-24:
Hi Tobias,
Sorry for the late reply. So the my definitions are :
#define ENABLE_STATE_SAVE_SPIFFS        
//#define ENABLE_STATE_SAVE_EEPROM    

When i restart ESP it continues with the last mode.
The output of http:///esp_status is what i see when i click on esp_status under commands in app?
My settings in arduino IDE shown on screenshot below.
[image]

dimitris17563, 2018-09-19:
Hi Tobias,
I was many days abroad and i couldnt test the device but now i am back and i looked again. Finally i found what was the problem. I did what you told me to check what triggers that lines SET MODE and i noticed that there were a strange behavior when i had opened and connected the adroid app. The lights were flickering sometimes or stack and didnt do the mode normal, the speed slider didnt had a smooth control etc. When i closed the app the device was working fine , modes fine, sliders fine. So i unistalled the app and install again (maybe i had old version?) and now it is working perfect! 
Thank you very much for your help and congrats for your nice work!

Best regards 
Dimitris

toblum, 2018-09-19:
Ah OK, that explains it!
Great that you found it out!

Regards
Tobias

closed 2018-09-19 by toblum