toblum/McLighting #216: Strip displaying wrong colors when using low brightness and NeoAnimationFX

Nobodi99, 2018-08-14:
Hello :)

I tested around with this and got some earlier troubles I had resolved.

Now I'm standing infront of a new problem.

When I use the WS2812FX library everything gets displayed nicely and when I change the brightness it just gets dimmer/brighter.

But when I use the NeoAnimationFX library I only get the color I want when I use maximum brightness.
When I lower the brightness the strip starts to get funky and displays random colors.

I have only HTTP_OTA and LEGACY_ANIMATIONS activated in the definitions.
And I use the index2.htm file if that is helpful.

Does anybody have an idea what I could possibly do wrong?

If needed I can record a  small video of the behavior so you can see what I mean.

toblum, 2018-08-14:
Hi @Nobodi99,

I don't have a NeoAnimationFX setup available here to test it out in that combination, but I never heard of this behaviour before. Could you see what's written in the serial monitor when you dim down?

Does that happen also when you're using index.htm instead of index2.htm? And also when you're setting the brightness directly be REST?

@debsahu Do you have any idea?

Regards
Tobias

debsahu, 2018-08-14:
I will be away from my computers and setups for another 2 weeks. Still settling in my new job. I have not seen such behaviour, could you send the outputs of "http://<IP>/esp_status"?

Nobodi99, 2018-08-14:
So I tried the above things. 

1) Serial monitor output is completely fine. The same as when using WS2812FX library.

2) I can't use the index.htm somehow. That's why I use the index2.htm.
    I get this by accessing the index.htm...

[image]

3) When using /set_brightness?c=10 or any other value I still get the same problem

4) The output from /esp_status:

{"HOSTNAME":"McLighting01","version":"2.1.2","heap":24696,"sketch_size":438848,"free_sketch_space":2703360,"flash_chip_size":4194304,"flash_chip_real_size":4194304,"flash_chip_speed":40000000,"sdk_version":"2.2.1(cfd48f3)","core_version":"2_4_1","cpu_freq":80,"chip_id":1458415,"animation_lib":"NeoAnimationFX","pin":"Ignored, check NEOMETHOD","number_leds":30,"button_mode":"OFF","home_assistant":"OFF","legacy_animations":"ON","esp8266_http_updateserver":"ON","state_save":"SPIFFS"}

toblum, 2018-08-14:
Hi @Nobodi99,

please use this index.htm:
https://github.com/toblum/McLighting/blob/master/clients/web/build/index.htm
The other one is just from a ESP example. Maybe I should delete this.

But as th issue is also with direct REST calls this seems not to be related with the UI.

What kind of ESP and strip do you use?

Tobias

debsahu, 2018-08-14:
could you try `/set_mode?c=10` ?

Nobodi99, 2018-08-14:
so using the correct index.htm is still producing the same result.

I'm using a NodeMcu v2 board with some WS2812 RGB leds. 

@debsahu I tried your command but it retains some of the leds in a very dim color but doesn't do anything else. The amount and brightness it retains is also random. Had just one Led but a bit brighter and also the whole strip but very dim.

debsahu, 2018-08-15:
Which pin are you using? NeoAnimationFX can only be used with RX pin using DMA method (default). Some boards have this pin with a pull down or up. In that case I will use UART method (`#define NEOMETHOD NeoPBBGRBU800`) for WS2812FX. This method uses pin D4.

Nobodi99, 2018-08-15:
I'm currently using the RX pin for trying both. NeoAnimationFX and WS2812FX. How can I test if it is a pullup or pulldown resistor?

Nobodi99, 2018-08-15:
Alright I guess I found the problem. 
I had a resistor of ~420 ohm  connected between the rx pin and the led input. 
I did that because it was recommended in some stuff I read about the leds. 

After having a look at the schematics today I saw that the NodeMcu v2 board already has a 470 ohm resistor connected to its RX pin on board. So I guess connecting another resistor was just too much for the signal to go through?

But still I'm confused that it works with bitbanging but not with DMA....
I guess the DMA output is a bit less stable/ delivers less current?

closed 2018-08-15 by Nobodi99