toblum/McLighting #217: Issues with WiFi connection

masterfink, 2018-08-15:
Hi there,

I have gone through the software installation instructions, and the board flashes just fine, but when I connect it to power and try to connect to the WiFi Manager, there is no access point. I've waited about 30 minutes each time to make sure, but it never shows up. 

I am using the same versions of the libraries mentioned in the software installation section. I have set the hostname in definiations.h. I've tried flashing on a Windows and Linux machine. I get the same result each time. Is there anything else that I can check?

masterfink, 2018-08-30:
@toblum yes I'll look at it tonight. I'm using an ESP8266 v2 board that was being used for another project that connected to wifi the "old way". I just got some new ESP8266 v3 boards (the larger sized ones) and can try one of those tonight as well.

sujitrp, 2018-09-11:
Use following tutorial 
1st erase flash
Then flash bin
http://cordobo.com/2300-flash-esp8266-01-with-arduino-uno/#more-2300

debsahu, 2018-09-24:
@masterfink Erase your flash completely when using a fresh ESP8266 or ones loaded with other firmware. Also provide the output of serial monitor, where it shows output from WiFiManager part of the code.

toblum, 2018-12-16:
Closing issue due to inactivity.

closed 2018-12-16 by toblum