toblum/McLighting #218: Upload on an ESP-12

dokho33, 2018-08-18:
Hi !
First sorry for my bad english :)

And this is my question: is it possible to upload sketch on a "simple" ESP-12 (not a nodemcu) ?

I'm trying to do this since yesterday but it doesn't work :(

This is what i can see in console: https://pastebin.com/zXRWhy64

(i don't paste it here because it's very long)

Someone have an idea ?

Regards,

toblum, 2018-08-19:
Dear @dokho33,

yes, a "pure" ESP12 is also possible. Some people use it also with a ESP01, but memory a bit bit limited then. 
The message looks like a issue with power supply. Make sure that it's stable and sufficient. Adding a 1000uF capacitor helps often.

Best regards
Tobias

dokho33, 2018-08-20:
Hi!
I don't have 1000µF so i've try with a 100µF and it's work. That is surprising it's that work without capacitor with an other sketch ...

The capacitor is needed when ESP is powered by fdti, in "normal" mode (powered by 5VCC power supply and voltage reductor) it's work without capacitor.

Thank you for help, now my 490 LEDs ribbon works like a charm.

Regards,

(sorry again for my english)

toblum, 2018-08-20:
HI @dokho33,

great. The capacitor is just to smooth the current spikes of the ESP that lead otherwise to a reboot. Every sketch is different so the current consumption of the ESP may also differ. 

I'll close the ticket.

Regards
Tobias

closed 2018-08-20 by toblum