toblum/McLighting #219: Problem with spaces in HOSTNAME and HomeAssistant

Mistercl0ud, 2018-08-20:
Hello,

Thank you for the time you put in that awesome code. Everything works great for me for the most part.
I have notice that if I tried to put a space in the hostname, MQTT auto discovery on HomeAssistant doesn't work as the device doesn't show up. As soon a I remove the space, everything works again. Any idea on how to fix that ?
HomeAssistant doesn't allow the change of a device's name if it was discovered with MQTT.
Thanks for the help !

toblum, 2018-08-22:
Hi @Mistercl0ud,

unfortunately I don't have much experience with Homassistant, maybe @debsahu has an idea when he finds time...

Best 
Tobias

debsahu, 2018-09-24:
Greetings @toblum I'm back.

@Mistercl0ud Its great that we found the limitations of MQTT topic names handled by HA. The hostname is used in the topic that is published, I would use "_" instead of spaces for seperation. As far as I see, this issues is on the HA's end.

closed 2018-09-24 by debsahu

toblum, 2018-09-27:
Hi @debsahu, great to have you back! I was a bit lost with the homeassistant questions in the meanwhile.