toblum/McLighting #220: How to change color for a effect by using the color wheel?

xeetdp, 2018-08-22:
Hi toblum, 

first of all I want to say thank you for the great project. 

I would like to change the color on a effect by using the color wheel. The color change by clicking the wheel will remove the effect and only change the color. I tried to edit the .js file but without a result.

cjs1502, 2018-08-22:
Choose the color first then the effect.

toblum, 2018-08-22:
Hi @xeetdp,

yes, this is the way the UI works at the moment. 
@cjs1502 : Good idea!

Maybe you want to try out the new "next-gen" UI: https://github.com/toblum/McLightingUI
Just upload the file as index2.htm and call it.

Regards
Tobias

closed 2018-08-22 by toblum

xeetdp, 2018-08-22:
@cjs1502 Thanks for the quick answer. But I want more dynamic. When I want change the effect from green to red the effect will disappear so I have to go back and click again. 

@toblum Also thanks for the response. The new UI looks exactly what I was searching for, but I use more then one stripe at the moment. What I did in the old version is to build a submenu and send ws command to differend IPs by using more then one JS files. 

For the index.html I use more then one implementation. Like this: 

`<script type="text/javascript" src="a1.js"></script>
 <script type="text/javascript" src="a2.js"></script>
`
I guess it's a stupid way but it's works fine for me so far and I'm a noob in JS and html. 

Regards 
Max

toblum, 2018-08-22:
Hi!

The new one has an experimental feature that allows you to enter multiple IPs in the settings to control more than one strip.

Tobias