toblum/McLighting #224: Brightness change causes reboot

tnkaiser, 2018-08-27:
If this has been discussed previously, please point me in the direction.

//#define USE_NEOANIMATIONFX  // Uses NeoAnimationFX, PIN is ignored & set to RX/GPIO3 or UART method: D4/GPIO2, see: https://github.com/debsahu/NeoAnimationFX
#define USE_WS2812FX          // Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX

// Neopixel
#define PIN D1           // PIN (14 / D5) where neopixel / WS2811 strip is attached
#define NUMLEDS 300       // Number of leds in the strip
#define BUILTIN_LED 2    // ESP-12F has the built in LED on GPIO2, see https://github.com/esp8266/Arduino/issues/2192
#define BUTTON 4         // Input pin (4 / D2) for switching the LED strip on / off, connect this PIN to ground to trigger button.

const char HOSTNAME[] = "McLighting01";   // Friedly hostname

#define HTTP_OTA             // If defined, enable ESP8266HTTPUpdateServer OTA code.
//#define ENABLE_OTA         // If defined, enable Arduino OTA code.
#define ENABLE_AMQTT         // If defined, enable Async MQTT code, see: https://github.com/marvinroger/async-mqtt-client
//#define ENABLE_MQTT        // If defined, enable MQTT client code, see: https://github.com/toblum/McLighting/wiki/MQTT-API
#define ENABLE_HOMEASSISTANT // If defined, enable Homeassistant integration, ENABLE_MQTT must be active
#define ENABLE_BUTTON        // If defined, enable button handling code, see: https://github.com/toblum/McLighting/wiki/Button-control
#define MQTT_HOME_ASSISTANT_SUPPORT // If defined, use AMQTT and select Tools -> IwIP Variant -> Higher Bandwidth
//#define ENABLE_LEGACY_ANIMATIONS
***************************************************************************************************
MQTT: Recieved [home/McLighting01_ha/state/in]: {"state": "ON", "brightness": 132}

Exception (28):
epc1=0x40227e7e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3fffea70 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffec10:  3fff4592 00000000 00000020 00000000  
3fffec20:  3fff4595 3fffecd0 3fffeca0 4020b458  
3fffec30:  00000004 00000084 3ffe9a48 3fff4595  
3fffec40:  3fff55c0 00000008 3fff458c 40228124  
3fffec50:  00000000 3fffecd0 00000008 00000026  
3fffec60:  3fff1c28 00000008 3fffec80 3fff45ae  
3fffec70:  00000000 3ffe8308 3fffeca0 4020b181  
3fffec80:  7b203a5d 61747322 3a226574 3fff45ae  
3fffec90:  3fff1c28 3fff4592 3ffe95a8 3fff45ae ........
................................................................................
fffff30:  3fff1338 000000b4 40222d54 3fffefb0  
3fffff40:  3fffdad0 00000009 00000000 40201593  
3fffff50:  00000000 3fff0f7c 3fff1068 4021a7ec  
3fffff60:  1b02089e 00000000 3fff135c 40214625  
3fffff70:  40224fb8 00000000 00001388 3fff1d00  
3fffff80:  3fffdad0 00000000 3fff0f7c 4021a81d  
3fffff90:  3fffdad0 00000000 3fff1cfa 4020f761  
3fffffa0:  00000000 00000000 00000001 3fff1d00  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

toblum, 2018-08-29:
HI @tnkaiser,

I think this wasn't reported before. Does this happen also when you set the brightness via web UI? Oder directly via REST call? Or only via MQTT call from homeassist?

Regards
Tobias

tnkaiser, 2018-08-30:
AMQTT ---> reboot  (via the popup slider when the light is turned on)
Direct MQTT --->reboot (via third party MQTT client), but only sporadically (sometimes, when the JSON string is sent WITHOUT anything else... i.e.  {"brightness":25} it works, other times it hangs
WebUI --->works, no reboot
websockets --->seems to work, no reboot.

toblum, 2018-08-30:
OK, then this seems to be related to MQTT. Strange that it happens fot both MQTT and AMQTT.
What kind of board do you use?
And what is returned when you call http://YOUR_IP/esp_status 

Regards
Tobias

MNatzke, 2018-09-25:
Hi all,

As you are waiting for an answer for about a month and I am having the same issue this is what I can provide for an analysis:
{"HOSTNAME":"BadWaschtisch","version":"2.1.4","heap":23504,"sketch_size":449056,"free_sketch_space":2695168,"flash_chip_size":4194304,"flash_chip_real_size":4194304,"flash_chip_speed":40000000,"sdk_version":"2.2.1(cfd48f3)","core_version":"2_4_2","cpu_freq":80,"chip_id":1458208,"animation_lib":"WS2812FX","pin":4,"number_leds":100,"button_mode":"ON","amqtt":"ON","home_assistant":"ON","legacy_animations":"ON","esp8266_http_updateserver":"ON","arduino_ota":"ON","state_save":"SPIFFS"}

Maybe on thing more:
If I set the brightness via "normal" MQTT Interface. 
```
BadWaschtisch/in
%100
```
There will is no reboot. It looks like it is always if you are using the Topics for "Home Assistant".

Best
Michael

toblum, 2018-09-27:
@MNatzke Thank you for your investigations.

@debsahu I couldn't reproduce this. Do you have an idea why this seems to only happen with the homeassistant topics?

MrPaulAR, 2018-09-28:
I'll add that I just installed this on a D1 Mini and I'm having the same issue with the AMQTT library.  Even sending a message with only the brightness causes core dump.

{ 
"brightness":25
}

debsahu, 2018-09-29:
Don't have any idea why there is a issue with JSON parsing. I'll investigate!

mkova186, 2018-10-01:
Shame issue with AMQTT and HA

{"HOSTNAME":"McLighting01","version":"2.1.4","heap":26432,"sketch_size":438064,"free_sketch_space":2707456,"flash_chip_size":4194304,"flash_chip_real_size":4194304,"flash_chip_speed":40000000,"sdk_version":"2.2.1(cfd48f3)","core_version":"2_4_2","cpu_freq":80,"chip_id":1458376,"animation_lib":"WS2812FX","pin":14,"number_leds":136,"button_mode":"ON","amqtt":"ON","home_assistant":"ON","legacy_animations":"ON","esp8266_http_updateserver":"ON","state_save":"SPIFFS"}

debsahu, 2018-10-02:
@MNatzke @MrPaulAR @mkova186 could you anyone try the develop version and update arduinojson to latest(6.4.0-beta) as well?

MNatzke, 2018-10-02:
I am on a trip with my family starting tomorrow. If I could not manage to test it tonight i'll test it in a week.
Will give you a feedback asap

mkova186, 2018-10-02:
same here, already had Arduinojson installed in 6.4.0-beta

mkova186, 2018-10-02:
just downgrade the Arduinojson to 6.1.0-beta and no issue, all working as it should. Tnx

BananaPukeh, 2018-10-02:
Same here, downgraded to 6.1.0-beta and it's all working fine. Thanks for figuring out

MNatzke, 2018-10-02:
Just tested the development version with 6.4.0-beta and YES everything is working very well. 
arduionojson 6.4.0-beta wit the master branch causes a crash once i try to change the brightness.

Thank you!

commit 6d8269f9 (2018-10-02): fix #224

- try to fix the way brightness is processed
commit 1821f39b (2018-10-02): Merge pull request #242 from toblum/develop

fix #224
closed 2018-10-02 by debsahu