toblum/McLighting #233: web client does not appear

mentaluproar, 2018-09-12:
I could not get it to upload the upload page, so I flashed it over via the Arduino plugin.  The ESP does connect to my network, the router shows it fetched an IP address, but it doesn't seem to want to host any web page.  What should I do to get the webUI working?

mentaluproar, 2018-09-12:
okay, so hostname.local DOES show something, but its the "ESP monitor."  Nothing lighting related.  Nothing pops up when accessed via IP address.

mentaluproar, 2018-09-12:
Found the data folder had files by the right name but they had different content from what's listed in the directions.

Test:  that did it.  I was struggling with this for two days and figured it out right after posting here.  Sorry about that.

toblum, 2018-09-12:
No problem! Great, that you found it out yourself.

Regards
Tobias

closed 2018-09-12 by toblum