toblum/McLighting #234: Alexa Voice support for McLighting

sujitrp, 2018-09-14:
Hi Using Openhab skill on alexa  & google home Power On off functionality is working good . 
but color is only working on google home . Color Item is not working with alexa .

Any suggestion ?

[image]

toblum, 2018-09-16:
Hi @sujitrp,

I have no experience with openhab. How is McLighting connected? Is this problem related to McL or more to openhab? I this latter is true, this might not be the best place to get help.

Regards
Tobias

toblum, 2018-09-19:
Great! This seems to work quite good. 

Regards
Tobias

sujitrp, 2018-09-22:
Now the next target is to remove Raspberry Pi and Openhab from set up setup something like 
[image]

sujitrp, 2018-09-24:
Hi Need help 
in above approach I am facing following raised issue . 

https://github.com/toblum/McLighting/issues/185#issuecomment-424026531

sujitrp, 2018-09-24:
Hi Finally i am connected to Io.adafruit.com by changing
line 352 to 'strcpy(mqtt_port, custom_mqtt_port.getValue());'
now next issue MQTT connection gets disconnected frequently .. Please find the log

MQTT active: io.adafruit.com:1883
Open http://192.168.1.28/ to open McLighting.
Use http://openhab/feeds.local/ when you have Bonjour installed.
New users: Open http://192.168.1.28/upload to upload the webpages first.

Couldn't find "/stripstate.json"
Failure!
readEEPROM(): STA| 1| 11|196|196|255| 0| 0
Found saved state: STA| 1| 11|196|196|255| 0| 0
ws2812fx_mode: 11
ws2812fx_speed: 196
brightness: 196
main_color.red: 255
main_color.green: 0
main_color.blue: 0
*WM: freeing allocated params!
MQTT disconnected, reconnecting!
Attempting MQTT connection 1 / 4 ...
MQTT connected!
MQTT topic in: openhab/feeds/in
MQTT topic out: openhab/feeds/out
MQTT: Message arrived [0100]
MQTT: Message arrived [0100]
MQTT: Message arrived [off]
MQTT: Message arrived [/red]
MQTT: Set WS2812 mode: [/red]
SET_MODE: 0 0
Saving cfg: {"mode":1,"strip_mode":0,"brightness":196,"speed":196,"red":255,"green":0,"blue":0} Success!
MQTT: Send [home/openhab/feeds_ha/state/out]: {"state":"ON","color":{"r":255,"g":0,"b":0},"brightness":196,"color_temp":327,"speed":196,"effect":"Static"}
Heap size: 25264
writeEEPROM(): STA| 1| 0|196|196|255| 0| 0
MQTT disconnected, reconnecting!
Attempting MQTT connection 2 / 4 ...
MQTT connected!
MQTT topic in: openhab/feeds/in
MQTT topic out: openhab/feeds/out
MQTT: Message arrived [=ff0000]
MQTT: Activated mode [1]!
Saving cfg: {"mode":1,"strip_mode":0,"brightness":196,"speed":196,"red":255,"green":0,"blue":0} Success!
MQTT: Send [home/openhab/feeds_ha/state/out]: {"state":"ON","color":{"r":255,"g":0,"b":0},"brightness":196,"color_temp":327,"speed":196,"effect":"Static"}
Heap size: 24888
MQTT disconnected, reconnecting!
Attempting MQTT connection 3 / 4 ...
MQTT connected!
MQTT topic in: openhab/feeds/in
MQTT topic out: openhab/feeds/out
MQTT: Message arrived [#ff0000]
MQTT: Set main color to: R: [255] G: [0] B: [0]
Saving cfg: {"mode":1,"strip_mode":0,"brightness":196,"speed":196,"red":255,"green":0,"blue":0} Success!
MQTT: Send [home/openhab/feeds_ha/state/out]: {"state":"ON","color":{"r":255,"g":0,"b":0},"brightness":196,"color_temp":327,"speed":196,"effect":"Static"}
Heap size: 24848
MQTT disconnected, reconnecting!
Attempting MQTT connection 4 / 4 ...
MQTT connected!
MQTT topic in: openhab/feeds/in
MQTT topic out: openhab/feeds/out
MQTT connection failed, giving up after 4 tries ...
sd

debsahu, 2018-09-24:
@sujitrp I dont think this is a McLighting related issue, is it? Its your very specific case where you are trying to communicate with Adafruit.io which needs a special code and there are limitations on bandwidth and number of messages you can send to adafruit.io. I suggest running your own broker like mosquitto or buying cloud instances of mosquitto broker.

toblum, 2018-09-27:
@sujitrp I agree with @debsahu 
Adafruit.io MQTT seems to behave different than other MQTT implementations. There are a lot of issues out there that are related to this (e.g. https://github.com/knolleary/pubsubclient/issues/291). I'm afraid this isn't supported with McLighting at the moment.

closed 2018-09-27 by toblum

sujitrp, 2018-09-28:
I am almost close to achive this . For me adafruit mqtt connected using pubsublib.  It passing on message in & out both. 
Adafruit is updating its dashboard  as per MCL output (e.g. ok 139%).
Now next step colour code conversion to her. E.g. red  to #ff0000.
Being not a programmer it is taking time for me to read try and do. 
One of my friend created colour code change  web service on Google cloud for me.  
Target is to open voice control .