toblum/McLighting #236: Issue with more rgb stripes

Hackerboy47, 2018-09-16:
Hello together, 
is it possible that you can get problems with the controling of the rgb, if you place several of them parallel? 
It works but sometimes it affects the rgbs.. 
I also put for every rgb signal an level shifter in front to amplify the signal. 
Has anyone some experience with that issue? 
Thanks a lot in advance

toblum, 2018-09-16:
Hi @Hackerboy47,

I have no experience with many strips in parallel, as this is no standard. This may work if you're lucky but it's not the mmost reliable option. The signal quality of the data line get's worse if you use more than one strip. a level shifter may help.
Maybe other users have experiences with that kind of setup. Maybe you could also ask at https://gitter.im/mclighting/Lobby

Regards
Tobias

Hackerboy47, 2018-09-16:
Hi @toblum,
Thanks for the advice. 
By the way, its really a great project :)

toblum, 2018-09-16:
@Hackerboy47 

Great! Good to hear!

closed 2018-09-16 by toblum