toblum/McLighting #237: Can we make code work offline with hotspot ?

savaliyadnk, 2018-09-17:


ashishagravat, 2018-09-17:
Hello Sir 
can we make this project without internat connection because of the web interface need internate.
i am to modification on code and Make  esp as a router all working fine but html page not see .
for the visible web interface need to  internate connection right?
can i downlord full web page offline   ? and upload on esp so i can run without internet connection

toblum, 2018-09-19:
HI @savaliyadnk,

this is not so easy to accomplish as everything must be served locally. There was another attempt to do this here: https://github.com/toblum/McLighting/issues/140 but I found it not to work reliably in the whole project context with to local AP enabled because of issues with the webserver library. So this functionality is not available for now. As a workaround you can user a local hotspot from your phone to make it work mobile.

Regards
Tobias

closed 2018-09-19 by toblum

BloodADM, 2018-10-15:
> Hello Sir
> can we make this project without internat connection because of the web interface need internate.
> i am to modification on code and Make esp as a router all working fine but html page not [...]

in my repository there is moddified index.html.cpp with js implemented in html to make it work offline without internet

kotelmach, 2018-10-17:
I actually made this modification, and it was working great, however there is now a small error in the code.  Would you be willing to see if you can fix it?

BloodADM, 2018-10-18:
Np :), what kind of errors ?