toblum/McLighting #238: Just 1 LED ist working

djamplifier, 2018-09-22:
Hallo,

Ich habe den Sketch fehlerfrei kompiliert. Leider wird bei meinem ws2811 led strip nur die erste LED angesteuert. Ich nutze einen Wemos D1 an Pin D4. Ich habe schon einen anderen Pin probiert, kein unterschied. Mit espeasy und Neopixel Fx plugin funktioniert es super. NUMLEDS ist gesetzt. Auch die Max anzahl von 100 leds habe ich beachtet. woran kann es liegen?

toblum, 2018-09-24:
Hallo @djamplifier,

das ist ungewöhnlich. Welche Ausgabe bekommst du denn, wenn du http://IP_ADRESSE/esp_status aufrufst? Wie steuerst du McLighting? Per Weboberfläche? Was passiert, wenn du http://IP_ADRESSE/set_mode?m=0&rgb=FF0000 direkt aufrufst?

Gruß
Tobias

---------------------------------------------------------

Hello @djamplifier,

this is unusual. What output do you get when you call http://IP_ADRESSE/esp_status? How do you control McLighting? Via web interface? What happens if you call http://IP_ADRESSE/set_mode?m=0&rgb=FF0000 directly?

Greetings
Tobias

djamplifier, 2018-09-25:
Hallo, ich habe es per weboberfläche gesteuert. Jetzt habe ich es nachdem ich auf neoanimationfx umgestellt habe und den RX pin verwende hinbekommen. Integration in Openhab2 über mqtt ist auch angeschlossen.  Danke! Super Programm!

toblum, 2018-09-27:
Hallo!

Freut mich, dass es nun läuft!

closed 2018-09-27 by toblum