toblum/McLighting #241: openhab - mqtt

bagotu, 2018-10-02:
Hello. Can't get a bunch to work "openhab - McLighting. Where can I read in detail about this? Can there are examples?
It did not help:
https://community.openhab.org/t/creating-controls-for-mclighting-project/42368/6

MNatzke, 2018-10-02:
Can you explain your problem in more detail?
I understood that you tried to add a control for in OpenHAB for McLightning and something does not work.
Where you able to find out if you have an issue with OpenHAB or with McLightning?

1) can you access McLighning via the WebInterface
2) are you able to accesss McLighning manually via (MQTT, WebSocket or any other API described in the Wiki of McLightning
3) Have you tried to use a MQTT Client on your computer to listen whats hapening in the Topics for McLightning

And please describe what is working and what not

Best
Michael

bagotu, 2018-10-02:
I do not know how to properly register everything.
for example, the dimmer is regulated via the web McLightning. and I see these changes on the web openHab. but on the turnover there is no change.
from openhub requests go. I watched here, MQTT.fx

here, http://192.168.1.66/index.htm - fully controlled

`Group Ledstrip

Switch Clrlight01power "Temple light" (Ledstrip) 
{mqtt= ">[mosquitto:McLighting01/in:command:ON: =all],
        >[mosquitto:McLighting01/in:command:OFF: =off],
        <[mosquitto:McLighting01/out:state:ON:OK =all],
        <[mosquitto:McLighting01/out:state:OFF:OK =off]"}

Dimmer CLRlight01Dim "Temple light Brightness [%d]" (Ledstrip) 
{mqtt= ">[mosquitto:McLighting01/in:command:*:JS(scale100-255.js)],
        <[mosquitto:McLighting01/out:state:JS(scale255-100.js)]", autoupdate="false"}
/*
Dimmer CLRlight01Speed "Temple light Speed [%.0f]" (Ledstrip) 
{mqtt= ">[mosquitto:McLighting01/in:command:*:JS(speedscale100-255.js)],
        <[mosquitto:McLighting01/out:state:JS(speedscale255-100.js)]", autoupdate="false"}
*/
Color Clrlight01 "Temple light" (Ledstrip) 
{mqtt= ">[mosquitto:McLighting01/in:command:*:JS(colour.js)]"}

Number Coloreffects "Temple Effects" (Ledstrip) 
{mqtt= ">[mosquitto:McLighting01/in:command:*:MAP(ColoreffectsJSON.map)]"}`

this is my first use MQTT, sorry

debsahu, 2018-10-02:
@MNatzke Once you guys figure this out, lets put a working version in clients folder for future reference. And could you also add some useful info in the wiki as well? Many thanks in advance. I dont have OpenHAB setup as I migrated to HA a few years ago.

sujitrp, 2018-10-02:
Hi bagotu
You are taking about my project . 
Now it is late Nightingale India. U will post ur questions or problem I should answer them . Cu tomorrow ...

bagotu, 2018-10-03:
> if you Change the Settings in OpenHAB you do not see the Changes in the McLighning Web Interface

It's true!

sujitrp, 2018-10-04:
to resolve this issue "if you Change the Settings in OpenHAB you do not see the Changes in the McLighning Web Interface"

use following rule in openhab for each item : 

rule PublishOnGXRINGpowerChangeOnly
when
Item GXRINGpower changed
then
Thread::sleep(100)
publish(“mqtt”,“XRING01/in”, transform(“MAP”, “ColoreffectsJSON.map”, GXRINGpower.state.toString))
publish(“mqtt”,“XRING02/in”, transform(“MAP”, “ColoreffectsJSON.map”, GXRINGpower.state.toString))
publish(“mqtt”,“XRING03/in”, transform(“MAP”, “ColoreffectsJSON.map”, GXRINGpower.state.toString))
publish(“mqtt”,“XRING04/in”, transform(“MAP”, “ColoreffectsJSON.map”, GXRINGpower.state.toString))
postUpdate(XRING01power, GXRINGpower.state.toString)
postUpdate(XRING02power, GXRINGpower.state.toString)
postUpdate(XRING03power, GXRINGpower.state.toString)
postUpdate(XRING04power, GXRINGpower.state.toString)
end

sujitrp, 2018-10-13:
Bagotu: you can close this as this is already solved in 
https://community.openhab.org/t/creating-controls-for-mclighting-project/42368/11

toblum, 2018-10-13:
Great!

closed 2018-10-13 by toblum