toblum/McLighting #245: Turn off BUILT IN LED

JoJ123, 2018-10-11:
How can I turn off the BUILTIN_LED?

debsahu, 2018-10-11:
Set it to an unused pin.

JoJ123, 2018-10-11:
I set the LED Strip to D4
So f.e. D1 is free, so I change this line:
`#define BUILTIN_LED D1`

But the blue light is still there

debsahu, 2018-10-11:
Try D6/GPIO12: `#define BUILTIN_LED 12` Or comment out all lines that have BUILTIN_LED in it.

JoJ123, 2018-10-12:
Both didn't turn the LED off. When I comment everything out, the LED is "blinking"

I have the D1Mini with the ESP8266MOD Module

debsahu, 2018-10-12:
Ha I figured it out, according to https://wiki.wemos.cc/products:d1:d1_mini your LED pin is on D4. Change the strip connection to another pin to have complete control over BUILTIN_LED. Closing, please reopen if this did not solve your issue.

closed 2018-10-12 by debsahu

nicoausnrw, 2020-09-16:
if the `BUILTIN_LED` set right than you can change in line 477
```
  digitalWrite(BUILTIN_LED, LOW);
```
to

```
  digitalWrite(BUILTIN_LED, HIGH);
```