toblum/McLighting #246: cant compile help

Ryanmaturi, 2018-10-14:
using the suggested versions of everything i get the following.
please help. 

Arduino: 1.8.1 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 4M (1M SPIFFS), ck, Disabled, None"

In file included from C:\Users\ryanm\Desktop\McLighting-master\McLighting-master\Arduino\McLighting\McLighting.ino:218:0:

sketch\request_handlers.h: In function 'String listStatusJSON()':

request_handlers.h:281: error: 'DynamicJsonDocument' was not declared in this scope

   DynamicJsonDocument jsonBuffer(bufferSize);

   ^

request_handlers.h:281: error: expected ';' before 'jsonBuffer'

   DynamicJsonDocument jsonBuffer(bufferSize);

                       ^

request_handlers.h:282: error: 'jsonBuffer' was not declared in this scope

   JsonObject root = jsonBuffer.to<JsonObject>();

                     ^

request_handlers.h:282: error: expected primary-expression before '>' token

   JsonObject root = jsonBuffer.to<JsonObject>();

                                             ^

request_handlers.h:282: error: expected primary-expression before ')' token

   JsonObject root = jsonBuffer.to<JsonObject>();

                                               ^

request_handlers.h:288: error: use of deleted function 'ArduinoJson::JsonArray::JsonArray(const ArduinoJson::JsonArray&)'

   JsonArray color = root.createNestedArray("color");

                                                   ^

In file included from C:\Users\ryanm\OneDrive\Documents\Arduino\libraries\ArduinoJson/include/ArduinoJson.hpp:11:0,

                 from C:\Users\ryanm\OneDrive\Documents\Arduino\libraries\ArduinoJson/include/ArduinoJson.h:8,

                 from C:\Users\ryanm\OneDrive\Documents\Arduino\libraries\ArduinoJson/ArduinoJson.h:8,

                 from C:\Users\ryanm\Desktop\McLighting-master\McLighting-master\Arduino\McLighting\McLighting.ino:47:

C:\Users\ryanm\OneDrive\Documents\Arduino\libraries\ArduinoJson/include/ArduinoJson/JsonArray.hpp:40:7: note: 'ArduinoJson::JsonArray::JsonArray(const ArduinoJson::JsonArray&)' is implicitly deleted because the default definition would be ill-formed:

 class JsonArray : public Internals::JsonPrintable<JsonArray>,

       ^

In file included from C:\Users\ryanm\OneDrive\Documents\Arduino\libraries\ArduinoJson/include/ArduinoJson/JsonArray.hpp:12:0,

                 from C:\Users\ryanm\OneDrive\Documents\Arduino\libraries\ArduinoJson/include/ArduinoJson.hpp:11,

                 from C:\Users\ryanm\OneDrive\Documents\Arduino\libraries\ArduinoJson/include/ArduinoJson.h:8,

                 from C:\Users\ryanm\OneDrive\Documents\Arduino\libraries\ArduinoJson/ArduinoJson.h:8,

                 from C:\Users\ryanm\Desktop\McLighting-master\McLighting-master\Arduino\McLighting\McLighting.ino:47:

C:\Users\ryanm\OneDrive\Documents\Arduino\libraries\ArduinoJson/include/ArduinoJson/Data/ReferenceType.hpp:31:3: error: 'ArduinoJson::Internals::ReferenceType::ReferenceType(const ArduinoJson::Internals::ReferenceType&)' is private
[... 689 lines cut ...]
                               ^

C:\Users\ryanm\Desktop\McLighting-master\McLighting-master\Arduino\McLighting\McLighting.ino: In lambda function:

McLighting:869: error: 'class Ticker' has no member named 'active'

       if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                        ^

McLighting:872: error: 'class Ticker' has no member named 'active'

       if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                             ^

Multiple libraries were found for "ArduinoJson.h"
 Used: C:\Users\ryanm\OneDrive\Documents\Arduino\libraries\ArduinoJson
 Not used: C:\Users\ryanm\OneDrive\Documents\Arduino\libraries\ArduinoJson-master
exit status 1
'DynamicJsonDocument' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

debsahu, 2018-10-14:
Please check https://github.com/toblum/McLighting/wiki/Software-installation for exact version of libraries needed for this project. From the errors I see you need ArduinoJSON v 6.4.0-beta.

TobiasJu, 2018-10-24:
I've got the exact same error as
I realized that i have installed: ArduinoJSON v 5.1.0
The newest v 6.4 is here: https://github.com/bblanchon/ArduinoJson/releases
I cloud not install it, because my Arduino IDE was too old (1.6.4) so i updated my IDE.
Than i updated all Libraries and set the Board Manager URL to: http://arduino.esp8266.com/stable/package_esp8266com_index.json
But now i get this Error Msg:  
  
`Arduino: 1.8.7 (Windows 10), Board: "WeMos D1 R2 & mini, 80 MHz, 115200, 4M (3M SPIFFS)"

In file included from C:\Users\Tobias\Documents\Arduino\McLighting-2.1.0\Arduino\McLighting\McLighting.ino:218:0:

sketch\request_handlers.h: In function 'void checkpayload(uint8_t*, bool, uint8_t)':

request_handlers.h:414:22: error: 'class Ticker' has no member named 'active'

     if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                      ^

request_handlers.h:417:29: error: 'class Ticker' has no member named 'active'

       if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                             ^

request_handlers.h:434:24: error: 'class Ticker' has no member named 'active'

       if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                        ^

request_handlers.h:464:24: error: 'class Ticker' has no member named 'active'

       if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                        ^

request_handlers.h:467:29: error: 'class Ticker' has no member named 'active'

       if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                             ^

request_handlers.h:489:24: error: 'class Ticker' has no member named 'active'

       if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                        ^

request_handlers.h:492:29: error: 'class Ticker' has no member named 'active'

       if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                             ^

request_handlers.h:572:26: error: 'class Ticker' has no member named 'active'

         if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                          ^

request_handlers.h:575:31: error: 'class Ticker' has no member named 'active'

         if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                               ^

request_handlers.h:645:24: error: 'class Ticker' has no member named 'active'

       if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);
[... 167 lines cut ...]
         if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                               ^

C:\Users\Tobia\Documents\Arduino\McLighting-2.1.0\Arduino\McLighting\McLighting.ino: In lambda function:

McLighting:872:24: error: 'class Ticker' has no member named 'active'

       if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);

                        ^

McLighting:875:29: error: 'class Ticker' has no member named 'active'

       if(!spiffs_save_state.active()) spiffs_save_state.once(3, tickerSpiffsSaveState);

                             ^

exit status 1
'class Ticker' has no member named 'active'

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.
`

debsahu, 2018-10-24:
Upgrade your esp8266 core library to > v2.4.1. ticker class was introduced in version 2.4.1 and above.

TobiasJu, 2018-10-24:
@debsahu thanks, I updated to v2.4.1. I forget to go to Tools>Boards>Boardmanager. This time i got this Error:   
  
Arduino: 1.8.7 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, 4M (3M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Build-Optionen wurden verändert, alles wird neu kompiliert
C:\Users\Tobias\Documents\Arduino\libraries\WS2812FX-master\WS2812FX.cpp: In member function 'void WS2812FX::mode_fade()':

C:\Users\Tobias\Documents\Arduino\libraries\WS2812FX-master\WS2812FX.cpp:397:41: error: no matching function for call to 'min(int, uint8_t&)'

   b = map(b, 0, 255, min(25, _brightness), _brightness);

                                         ^

C:\Users\Tobias\Documents\Arduino\libraries\WS2812FX-master\WS2812FX.cpp:397:41: note: candidates are:

In file included from c:\users\Tobias\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62:0,

                 from C:\Users\Tobias\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Arduino.h:255,

                 from C:\Users\Tobias\Documents\Arduino\libraries\WS2812FX-master\WS2812FX.cpp:53:

c:\users\Tobias\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4226:5: note: template<class _Tp, class _Compare> _Tp std::min(std::initializer_list<_Tp>, _Compare)

     min(initializer_list<_Tp> __l, _Compare __comp)

     ^

c:\users\Tobias\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4226:5: note:   template argument deduction/substitution failed:

C:\Users\Tobias\Documents\Arduino\libraries\WS2812FX-master\WS2812FX.cpp:397:41: note:   mismatched types 'std::initializer_list<_Tp>' and 'int'

   b = map(b, 0, 255, min(25, _brightness), _brightness);

                                         ^

In file included from c:\users\Tobias\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62:0,

                 from C:\Users\Tobias\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Arduino.h:255,

                 from C:\Users\Tobias\Documents\Arduino\libraries\WS2812FX-master\WS2812FX.cpp:53:

c:\users\Tobias\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4221:5: note: template<class _Tp> _Tp std::min(std::initializer_list<_Tp>)

     min(initializer_list<_Tp> __l)

     ^

c:\users\Tobias\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:4221:5: note:   template argument deduction/substitution failed:

C:\Users\Tobias\Documents\Arduino\libraries\WS2812FX-master\WS2812FX.cpp:397:41: note:   mismatched types 'std::initializer_list<_Tp>' and 'int'

   b = map(b, 0, 255, min(25, _brightness), _brightness);

                                         ^

In file included from c:\users\Tobias\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:61:0,

                 from C:\Users\Tobias\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Arduino.h:255,

                 from C:\Users\Tobias\Documents\Arduino\libraries\WS2812FX-master\WS2812FX.cpp:53:

c:\users\Tobias\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:239:5: note: template<class _Tp, class _Compare> const _Tp& std::min(const _Tp&, const _Tp&, _Compare)

     min(const _Tp& __a, const _Tp& __b, _Compare __comp)

     ^

c:\users\Tobias\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:239:5: note:   template argument deduction/substitution failed:

C:\Users\Tobias\Documents\Arduino\libraries\WS2812FX-master\WS2812FX.cpp:397:41: note:   deduced conflicting types for parameter 'const _Tp' ('int' and 'uint8_t {aka unsigned char}')

   b = map(b, 0, 255, min(25, _brightness), _brightness);

                                         ^

In file included from c:\users\Tobias\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:61:0,

                 from C:\Users\Tobias\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Arduino.h:255,

                 from C:\Users\Tobias\Documents\Arduino\libraries\WS2812FX-master\WS2812FX.cpp:53:

c:\users\Tobias\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:193:5: note: template<class _Tp> const _Tp& std::min(const _Tp&, const _Tp&)

     min(const _Tp& __a, const _Tp& __b)

     ^

c:\users\Tobias\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:193:5: note:   template argument deduction/substitution failed:

C:\Users\Tobias\Documents\Arduino\libraries\WS2812FX-master\WS2812FX.cpp:397:41: note:   deduced conflicting types for parameter 'const _Tp' ('int' and 'uint8_t {aka unsigned char}')

   b = map(b, 0, 255, min(25, _brightness), _brightness);

                                         ^

exit status 1
Fehler beim Kompilieren für das Board LOLIN(WEMOS) D1 R2 & mini.

debsahu, 2018-10-24:
Looks like you need to update WS2812FX. I would update all libraries described here: https://github.com/toblum/McLighting/wiki/Software-installation

debsahu, 2018-12-09:
Closing due to inactivity, you can reopen if required

closed 2018-12-09 by debsahu

rutujalole, 2020-02-10:
class ArduinoOTAMdnsClass<WiFiServer, WiFiClient, WiFiUDP>' has no member named 'onStart'
why this error is comes in to my laptop only
esp32 is installed