toblum/McLighting #247: Unable to overwrite Index.htm

Kurtisrc, 2018-10-14:
Hello, 
Would just like to start by saying that I am very much enjoying McLighting. It was exactly what I wanted for my Table and TV Lights. 
The issue I am having is that I am unable to overwrite the /index.html. Everytime I upload a newer version of it (Only make small changes so It's more personalised for me) it does not overwrite the original index.htm. 

A work around I have found is to upload it as index2.htm and change it in the "spiffs_webserver.h" and making the change below

bool handleFileRead(String path) {
	DBG_OUTPUT_PORT.println("handleFileRead: " + path);
	if (path.endsWith("/")) path += "index.htm";   **<---- Change this to index2.htm** 
	String contentType = getContentType(path);
	String pathWithGz = path + ".gz";
	if (SPIFFS.exists(pathWithGz) || SPIFFS.exists(path)) {
		if (SPIFFS.exists(pathWithGz))
			path += ".gz";
		File file = SPIFFS.open(path, "r");
		size_t sent = server.streamFile(file, contentType);
		file.close();
		return true;
	}
	return false;
}

Could you please advise me if there is something that I am doing wrong. 

The main reason why I would like to overwrite the index.htm is so I am able to upload your new McLightingUI and have that the main display. 

Thank you.

Kurtisrc, 2018-10-14:
Brilliant, Thank you for your quick reply. 

This has worked, soo thank you. Many hours of mind bobbling has been solved. 

Thank you.

closed 2018-10-14 by Kurtisrc

lukaty, 2018-10-14:
How about right click the file in browser and delete. Then upload new one.

Kurtisrc, 2018-10-14:
Unfortunately that does not work. I've tried so many different ways and the above that Debesahu has suggested has woks. 

I not sure why you can't just rename, delete and replace. Or just upload a fresh new one with the same name. 

But I now know what to do, so it's all good.

debsahu, 2018-10-14:
When baudrate is too high and one uses low quality cables, data corruption happens during partioning. This is known issue. Usually a clean erase fixes this. My mantra is use short reliable USB cable and initial partition done at low boud rates. Then I switch to 921k uploading sketch partition alone or use OTA.