toblum/McLighting #250: HA & MQTT with certificate?

Nobodi99, 2018-10-15:
Hello,

I have just finished setting up my homeassistant by using the hassbian image.
I used the mosquitto MQTT broker but I installed it so it uses a letsencrypt certificate for connecting.

Is there any way to implement that into McLighting so it can be used with the certificates or is there some workaround to use this without the certificates? 

I would rather not change the usage of the certificates because I have MQTT traffic from outside my network which I prefer to be certified.

Does anybody have an idea here?

debsahu, 2018-10-15:
You have to store the certificate in SPIFFs. See this gist: https://gist.github.com/eLement87/133cddc5bd0472daf5cb35a20bfd811e . Just a starting point, if someone figures it out let us know. I wanted to do this to connect to GCP and failed miserably.

Nobodi99, 2018-10-16:
Just a quick update and a question from me. 

I used the sample you posted and I was able to get all certificates onto the ESP and read all of them fine.
The problem I have now is that it still doesn't connect to the mqtt broker.

But I can't think of anything that's missing...

debsahu, 2018-10-16:
What is your TLS version for Mosquitto? 1.1?

Nobodi99, 2018-10-18:
Alright. 
So I played around with different settings, created dozens of certificates and tried more or less everything I could find to get this working...

In the process I found this code:
[https://github.com/mertenats/Open-Home-Automation/blob/master/openhome/sketches/Entrance/Entrance.ino](url)

It uses just the fingerprint of the certificate to connect. which saves space on the controller so you don't have to update the files every time your certificates update. Which letsencrypt does every 90 days...

For this you have to set `require_certificate false` in your `mosquitto.conf`

After testing it and finally getting it to work I will try to implement that into my local mclighting and see if I can get this to work.

And I'm forcing TLS v1 in mosquitto.

Nobodi99, 2018-10-18:
Will have a look at that later. maybe it's a nicer implementation. 

As soon as I had a working mqtt set up I made a backup of the full SD card from my pi.
It proved to be the best idea I had in this whole ordeal.
If anything goes wrong you can just roll back your setup and t will be as good as it was before messing around.
So if you start messing around make a backup first.

Nobodi99, 2018-10-20:
Quick update from my side. Tried to implement the code I mentioned earlier into an existing MCLighting code. 
My problem here is that I have no clue where I should add the verification of the certificate.
It needs to be added after the WiFi is connected but before the actual MQTT connection is established.
But my knowledge in Arduino is too shallow to find that part of the code...
Tried a few different spots but couldn't figure it out...

Could you point me to the part of the code where I would need to be?

At the moment I worked around the certificate by opening the unsecured MQTT port (1883) inside my home network. This works so far but it's not the final solution I'm opting for :/

debsahu, 2018-10-20:
Are you using async MQTT or PubSubClient?

Nobodi99, 2018-10-20:
Was trying to use the PubSubClient.

debsahu, 2018-10-20:
Define Wificlientsecure in Line 39 in McLighting.ino and necessary certificate check lines in 918-974 request_handlers.h

debsahu, 2018-10-23:
Closing this and following the same issue @ https://github.com/toblum/McLighting/issues/255

closed 2018-10-23 by debsahu