toblum/McLighting #252: Amperage Limit for LED Strips

Fusseldieb, 2018-10-18:
Hey,
I'd like to suggest a feature that's not yet present and that I've missed a ton. In FastLED's library you could set a amperage limit for the whole LED strip, and if the LED was using more than the specified amperage, it would lower the brightness a bit. That brightness change was so suttle that you don't even notice it.
For example: White would dim a bit to satisfy my 2A limit, but when I use red, it wouldn't dim at all, because it didn't exceed the specification.

On this project you use the NeoPixel library, which doesn't support that parameter and I've burned 3 Buck converters until now (and counting). With FastLED that wasn't a problem, there they were just fine.

sujitrp, 2018-10-19:
Not Sure on this but this might reduce consumption by 50 %

Line 626 - 627 

 server.on("/get_brightness", []() {
    String str_brightness = String((int) (brightness / 1.25));

toblum, 2018-10-20:
@Fusseldieb That sounds interesting, but this might better be implemented in the animation library and just used be McLighting.