toblum/McLighting #253: Multiple Custom scenes?

Sousanator, 2018-10-19:
I'm trying to add multiple custom scenes using WS2812FX but I'm not sure how to fill out the following parameters:

setCustomMode(const __FlashStringHelper* name, uint16_t (*p)()),

I got a single custom scene working using setCustomMode(myCustomScene), but I'm not sure what *p is to be able to add the other 4.

Any help?

toblum, 2018-10-20:
@Sousanator I never tried the custom modes of WS2812FX, but you may better ask in the WS2812FX project.

Regards
Tobias

debsahu, 2019-02-01:
It is a function that returns a delay in uint16_t format. See https://github.com/kitesurfer1404/WS2812FX/blob/077e3268944b59c156e704ad7f19f00b6239597d/examples/ws2812fx_custom_effect/ws2812fx_custom_effect.ino#L61

closed 2019-02-01 by debsahu