toblum/McLighting #254: SmartThings Integration

Sousanator, 2018-10-21:
McLighting works really well so I thought I'd make a SmartThings Device Handler for it.

I just wanted something quick and dirty to be able to control my light strips with Google Assistant so this just uses the basic HTTP command API (no json).  There's no true feedback or polling so changes from the web browser client don't get reflected here.  All the internal commands are available in webCore so you can do more advanced stuff in there.

This device handler can turn on/off the lights, set any color, set the brightness, change the scene, set the animation speed, and the ability to save and call all of the above with 6 different presets.

Here is the link to the device handler:
https://github.com/Sousanator/Smartthings-McLighting/blob/master/SmartthingsDevice_McLighting.groovy

[image]

[image]

debsahu, 2018-10-21:
This thing is fantastic, will surely try it. I was about to write something like this for ST.

sujitrp, 2018-10-22:
Hi 
 Few weeks back I used  McLightning with  Openhab - MQTT -Google Home / Alexa  
Openhab and MQTT on Raspberry Pi 3 B and McLightning  on WeMOS D1 Mini & WeMOS X Ring as LED. .
Set up worked fantastically without any Breakage . 
Next step is to reduce the hardware & Software  components . One area I Identified was Openhab -on software & Raspberry Pi 3 on hardware . Option I was thinking was adafruit.io with adafruit MQTT .
I can able to make On Off kind of functionality . 
Then I looked Mongoose-OS as a option .
But due to my  non coding capability I  have to keep it on hold . 
My requirement : Customisation in MQTT topic configuration on McLightning  
                             Colour conversion / Effect conversion.

debsahu, 2018-11-05:
@Sousanator I was able to add the device handler. Is there code for smartApp that is missing?

Sousanator, 2018-11-06:
I have thought about it, but don't have the time right now to implement it.
Since smartthings is the only way I control the lights it's not high on my priority list.

debsahu, 2018-11-06:
Do you want me to add this to clients and/or wiki?

evolwun, 2019-01-26:
> @Sousanator I was able to add the device handler. Is there code for smartApp that is missing? Nevermind, I was able to add a device and got it going! A feedback from ESP8266 would be better, have you thought of reading status from "http://<IP_Address>/status"

I must be missing the same thing you did, initially. (I am new to this stuff) I got the Device Handler but not the SmartApp... what did I miss?