toblum/McLighting #258: No connection to mqtt broker. Check/edit mqtt settings

gent99, 2018-10-26:
Hi,

i've got a problem to get my device connect to the mqtt broker. I'm using amqtt on the device. At home the connection works, at the office, with password protected mqtt server (mosquitto) i get 
```
Connecting to MQTT...
Disconnected from MQTT, reason: TCP Disconnected.
```
in serial monitor. 2 questions: does the amqtt library in mclighting support username and password? And is there a way to change the mqtt settings after the device is out of ap-mode?

Kind regards

debsahu, 2018-10-26:
Yes both MQTT and AMQTT support passwords. To reset settings, go to a web browser and type http://<IP_ADDRESS>/reset_wlan , where replace IP_ADDRESS with actual IP address.

gent99, 2018-10-26:
Thank you very much. It seems that it is an issue with our office network (unfortunately). But to know how to edit the settings is very helpful.

closed 2018-10-26 by gent99