toblum/McLighting #259: error compiling sketch

smartmac4, 2018-10-31:
when ever i try to compile the code for the esp8266 it gives me this error
i would like to get this working by tonight or tomorrow for Halloween any help would be great.

Arduino: 1.8.7 (Windows Store 1.8.15.0) (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\Micah\Documents\Arduino\libraries\async-mqtt-client-master\src/AsyncMqttClient.h:4:0,

                 from C:\Users\Micah\Desktop\McLighting-master\Arduino\McLighting\McLighting.ino:44:

C:\Users\Micah\Documents\Arduino\libraries\async-mqtt-client-master\src/AsyncMqttClient.hpp:11:25: fatal error: ESPAsyncTCP.h: No such file or directory

 #include <ESPAsyncTCP.h>

                         ^

compilation terminated.

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

smartmac4, 2018-10-31:
and i have disabled mqtt by the way if that effects anything

debsahu, 2018-10-31:
You have AMQTT enabled, which is async version of MQTT. All the required libraries are posted [here](https://github.com/toblum/McLighting/wiki/Software-installation)

smartmac4, 2018-10-31:
Thank you I was missing that libary works great! And another question is can I use it as a standalone device like just using it's own network that's being broadcasted by the esp?

debsahu, 2018-10-31:
Not possible at this moment, keep an eye out at https://github.com/toblum/McLightingUI for updates.

closed 2018-10-31 by debsahu