toblum/McLighting #260: Where would one add custom animations (utilizing the WS2812FX library)?

code-in-progress, 2018-10-31:
First off, GREAT project! It has literally saved me days of coding as I was thinking about something along the same lines project-wise.

With that said, I'm trying to port over some of my custom animations that I built with FastLED and I'm a bit stumped on where to add them in the code base. I'm thinking somewhere in request_handler.h, but I'm not entirely sure where.

Any thoughts?

toblum, 2018-11-01:
Hi @code-in-progress,

thank you for the good feedback.

You have to add new animation directly in the WS2812FX library as in this example. These effects should appear automatically in McLIghting:
https://github.com/kitesurfer1404/WS2812FX/blob/master/examples/ws2812fx_custom_FastLED/ws2812fx_custom_FastLED.ino

Best regards
Tobias

closed 2018-11-01 by toblum

code-in-progress, 2018-11-01:
Thanks @toblum! Much appreciated.