toblum/McLighting #268: Disconnected from MQTT, reason: MQTT not authorized.

hugofant, 2018-11-20:
Hi,
i'm acutally using the sketch version 2.1.5 and homeassistant 0.82.1 with Mosquitto broker addon 1.4.15.
My problem: since the last upgrade of the MosqittoAddon, McLighting is no longer able to connect anonymously - Other ESP8266s (sensors created with esphomeyaml) and also MQTT.fx are working well and are connecting anonymously without any problem....so it seems not to be a problem of the broker.
Serial Console says:
```
Connecting to MQTT...
Disconnected from MQTT, reason: MQTT not authorized.
Disconnected from MQTT, reason: TCP Disconnected.
```
and the broker:
```
1542721373: New connection from 10.10.0.14 on port 1883.
1542721373: Socket error on client <unknown>, disconnecting.
```
definitions.h:
```
const char HOSTNAME[] = "McLighting01";   // Friedly hostname

#define HTTP_OTA             // If defined, enable ESP8266HTTPUpdateServer OTA code.
//#define ENABLE_OTA         // If defined, enable Arduino OTA code.
#define ENABLE_AMQTT         // If defined, enable Async MQTT code, see: https://github.com/marvinroger/async-mqtt-client
//#define ENABLE_MQTT        // If defined, enable MQTT client code, see: https://github.com/toblum/McLighting/wiki/MQTT-API
#define ENABLE_HOMEASSISTANT // If defined, enable Homeassistant integration, ENABLE_MQTT must be active
#define ENABLE_BUTTON        // If defined, enable button handling code, see: https://github.com/toblum/McLighting/wiki/Button-control
#define MQTT_HOME_ASSISTANT_SUPPORT // If defined, use AMQTT and select Tools -> IwIP Variant -> Higher Bandwidth
#define ENABLE_LEGACY_ANIMATIONS
```
MQTT broker config:
```
{
  "logins": [],
  "anonymous": true,
  "customize": {
    "active": false,
    "folder": "mosquitto"
  },
  "certfile": "fullchain.pem",
  "keyfile": "privkey.pem"
}
```
would be thankfull for some hints/ideas

debsahu, 2018-11-20:
Looks like you just secured your MQTT connection, as of now MQTT cannot connect to SSL/TLS secured MQTT. Next update will have ability to connect to secured servers.

hugofant, 2018-11-20:
hmm, why do you think so - isn't port 1883 the unsecure one and mosquitto log also says:
```
[INFO] Setup mosquitto configuration
[WARN] SSL not enabled - No valid certs found!
[INFO] No local user available
[INFO] Initialize system configuration.
[INFO] Initialize Hass.io Add-on services
[INFO] Initialize Home Assistant discovery
[INFO] Start Mosquitto daemon
1542721341: mosquitto version 1.4.15 (build date 2018-03-04 15:35:59+0000) starting 
```

debsahu, 2018-11-20:
My assumption was that you have set hassio correctly with SSL certificates (because you had fullchain.pem and privkey.pem in your settings). Yes 1883 is insecure port for unencrypted communication. Is SSL is used, one would need to connect to port 8883.

I have always had issues with hassio, maybe try a minimal PubSubClient example and see if it works?

debsahu, 2018-11-20:
I re-read your first comment and see that you are able to use other devices with MQTT. Could you try commenting out this line in **McLighting.ino**
`//if (mqtt_user != "" or mqtt_pass != "") amqttClient.setCredentials(mqtt_user, mqtt_pass);`

hugofant, 2018-11-20:
great! works well with this modification....but a short explanation would be nice - as i haven't set the user/password.....why is/was it !="" ?
thanks a lot for your help

debsahu, 2018-11-20:
I'm guessing some value was either put in definitions.h for mqtt_user and/or mqtt_pass or during initial WiFiManager setup. Once that happens, the value is stored in json format on your SPIFFs. Upon reboot McLighting reads from SPIFFs and uses the stored value. In your case, the stored value might be incorrect.

closed 2018-11-20 by debsahu