toblum/McLighting #269: HomeKit Buttom

Camel2k, 2018-11-20:
Hallo Tobias,

wenn ich das ganze in HomeBridge einbinde, mit deiner Beispiel .json, bekomme ich einen Buttom in HomeKit in dem steht `derRaumname` "Nicht Unterstützt".
Das habe ich zu meiner Config hinzu gefügt:

                        "accessory": "HTTP-RGB",
                        "name": "Stripe",

                        "switch": {
                                "status": "http://192.168.178.83/get_switch",
                                "powerOn": "http://192.168.178.83/all?r=255&g=255&b=255",
                                "powerOff": "http://192.168.178.83/off"
                        },

                        "brightness": {
                                "status": "http://192.168.178.83/get_brightness",
                                "url": "http://192.168.178.83/set_brightness?c=%s"
                        },

                        "color": {
                                "status": "http://192.168.178.83/get_color",
                                "url": "http://192.168.178.83/all?rgb=%s",
                                "brightness": true
                        

Kannst Du mir da evtl auf die Sprünge helfen?
Gruß
Kai

toblum, 2018-11-20:
Hallo @Camel2k,

mit Homebridge habe ich schon einige Zeit nicht mehr gearbeitet. In dem aktuellen Konfigurationsbeispiel auf "https://www.npmjs.com/package/homebridge-better-http-rgb" wird noch "service": "Light", angegeben. Es kann sein, dass die Einstellung noch dazugekommen ist.
Findest du sonst noch Meldungen im Logfile von Homebridge?
----------------------------------------------------------------------------
I haven't worked with Homebridge in a while. In the current configuration example on "https://www.npmjs.com/package/homebridge-better-http-rgb" "service": "Light" is given. It is possible that the setting has been added.
Do you find any other messages in the logfile of Homebridge?
Viele Grüße
Tobias

toblum, 2018-11-20:
Freut mich, dass es nun klappt. Ich habe die Beispielconfig angepasst.

Viele Grüße

-------------------------------

I'm glad it's working now. I have modified the sample config.

Many greetings

closed 2018-11-20 by toblum