toblum/McLighting #271: Support for ESP32 Development board

Fisico, 2018-11-28:
i can't compile the sketch for my ESP32 board.
In Arduino Board selection it is called "ESP32 Dev Module"

Error:
McLighting:6:71: error: ESP8266WiFi.h: No such file or directory

compilation terminated.

exit status 1
ESP8266WiFi.h: No such file or directory

Wich Library i have to use in the scetch in order to get it working with the ESP32 Board? The ESP32 Board should not be that different than the board you have in your documentation.
https://github.com/espressif/arduino-esp32

toblum, 2018-11-29:
Hi @Fisico,

the sketch doesn't work on the ESP32 and for now we're not working on it. From software and hardware side the both ESP have some fundamental differences that are not obvious from the first sight.
But there is currently no need to use the more powerful and expensive ESP32 as it has no relevant advantages over the ESP8266 that handles the sketch without problems.

Regards
Tobias

closed 2018-11-29 by toblum

aenertia, 2018-12-19:
I feel like this is probably a worthwhile addition to see also. 

I have more esp32 boards and they are relatively cheaper in certain configurations than the esp01; so am not sure the expense argument is valid. 

On the surface looking through your sketch a lot of things should be abstracted like defines to make the core libraries used more modular. Perhaps a refactor with more general board support (not just esp32 but stm32 etc) is worth looking towards for a v3 release or something.