toblum/McLighting #275: 2813

QwertyInfo, 2018-12-03:


QwertyInfo, 2018-12-03:
Hi there! Just tried with WS2812 strip and all is workin as it should. 
Today i have a WS2813 4 pin Visdoll strip , 60led/m-5mt.
Ho do i connect Di/d0 (data return) to work?
Thank you

debsahu, 2018-12-03:
I dont have any WS2813 strips, but from what I could gather it should behave like WS2812B strips. The new chips have a backup data line which is labelled as DO, no connection is necessary. Your data line should be connected to DI. Again I got this info from the interwebs, dont know how accurate it is.

QwertyInfo, 2018-12-04:
> I dont have any WS2813 strips, but from what I could gather it should behave like WS2812B strips. The new chips have a backup data line which is labelled as DO, no connection is necessary. Your data [...]

Ok i'm investigating today ho to solve this.Tried connecting only D1,only D0 and i sometimes have only 1 led on at the end of the strip. Power consumption is 0.186mA,the same i have when connecting a WS8212 in IDLE mode.Strip is just unboxed

debsahu, 2018-12-04:
Terrible reviews of this on Amazon! WS2813 is not supported in Adafruit NeoPixel library that McLighting is based on. These are supported in FastLED and NeoPixelBus libraries, saying timing is like WS2811 and at 800kHz. Sorry wish could help you better, don't have these LEDs.

QwertyInfo, 2018-12-04:
Ok, I will try tomorrow. IMHO the ws2813 could be a very good strip, accordingly programmed. Let's try, brb

QwertyInfo, 2018-12-04:
Hey! I managed to light up about 21 leds from the beginning of the strip.
It's in a erratic behaviour,but now i know my strip is good ;)

Using NeoEsp8266Dma800KbpsMethod dma = NeoEsp8266Dma800KbpsMethod(NUMLEDS, 3)
"numleds" in definitions.h is 300 (60 each meter x5)

closed 2018-12-06 by debsahu

QwertyInfo, 2018-12-06:
Thanks for helping!