toblum/McLighting #276: API misbehavior

M4lik, 2018-12-03:
So, I'm not quite sure if this is a bug, or I'm just not getting it.
I am currently building my smart home system and wanted to integrate an esp (nodemcu) with McLighting.
Currently I'm going with MQTT (mabe I'm going to switch to HTTP because of more detailed features)
In the WS API doc (wich should be the same as the MQTT) it is explained that the `#<hexrgb> is setting a color and the `*<hexrgb>` is setting the color AND turning all lamps on

> Payload: `#<hexrgb>`
> Where <hexrgb> is the color as HEX, e.g. 04d2ff
> Set default color of the lamp.

> Payload: `*<hexrgb>`
> Where <hexrgb> is the color as HEX, e.g. 04d2ff
> Set default color of the lamp and light all LEDs in that color.

But this does not match with the behavior I'm experiencing. Although I call  `#<hexrgb>` my LED strip is turned on.

So again: Am I interpreting this wrongly, or is this a misbehavior?

MfG / Greets
Malik

debsahu, 2018-12-03:
The diff is: # sets only main color, so the animation can be anything, but the main color is changed without affecting the current mode. * sets the main color AND sets mode to static mode (old mode is replaced, mode=FX_MODE_STATIC).

M4lik, 2018-12-03:
Oh yeah, I see that I was definitly ucnlear what I meant. Sorry for that.
I forgot to mention that my strip is turned off. Then I set the color via `#<hexrgb>`. The problem now is that the strip turns on. I've done some very basic debugging via API and found out that when i set the color the `switch status` gets turned to 1.

> Here is some JSON
```JSON
{
  "mode": 2,
  "ws2812fx_mode": 0,
  "ws2812fx_mode_name": "Static",
  "speed": 196,
  "brightness": 196,
  "color": [
    0,
    250,
    212
  ]
}
```
Result of `http://myesp/status` after sending `=off` via MQTT
`http://myesp/get_switch` returns `0`

```JSON
{
  "mode": 1,
  "ws2812fx_mode": 0,
  "ws2812fx_mode_name": "Static",
  "speed": 196,
  "brightness": 196,
  "color": [
    255,
    0,
    0
  ]
}
```
Result of `http://myesp/status` after sending `#ff0000` via MQTT
`http://myesp/get_switch` returns `1`

This is a missbehavior as far as I can tell from a pov of a generic API (I send _one_ command and _two_ things change) and from the docs, since this should only be the case when `*<hexrgb>` is used.

Again I apologise for being unclear as heck what my problem is.

debsahu, 2018-12-03:
Ahh I see, https://github.com/toblum/McLighting/blob/7a44a76aca74d8501f6018c1d9dfc28a43ae45ee/Arduino/McLighting/McLighting.ino#L965 is to blame. The prevmode is never set back to HOLD, or any value other than SET_MODE unless brightness is called. Its a bug, thanks for reporting!

debsahu, 2018-12-03:
Upon second thought, # is the only way one can turn on McLighting. There is=off but not on. We turn on by setting main values for RGB. @toblum am I right?

closed 2018-12-08 by debsahu