toblum/McLighting #277: MQTT group topic when using HA

gent99, 2018-12-03:
Hi,
are there any plans that we get group topics with ha and amqtt enabled? We'd like to set 50+ McLighting devices to the same color and animation with just one mqtt payload. Anyone here willing to help me integrating this?

Kind regards, Holger

debsahu, 2018-12-03:
You would keep same hostname and uncomment ` //#define ENABLE_MQTT_HOSTNAME_CHIPID` in **definitions.h**

debsahu, 2018-12-03:
Firstly use https://github.com/toblum/McLighting/tree/develop where I recoded a few parts to set chipID as **mqtt_clientid**

This is a very specific case, but you have two options on the develop branch: 1. Use IP address directly, forget about MDNS 2. Make necessary changes to `bool mdns_result = MDNS.begin(HOSTNAME);` to `bool mdns_result = MDNS.begin(mqtt_clientid);`