toblum/McLighting #283: McLighting works until AMQTT or MQTT defined (with AMQTT or MQTT only till next boot)

SmilyHM, 2018-12-09:
**DE**

Hallo Tobias,
Hallo zusammen,

ich starte erst einmal auf DE, da mir das einfacher fällt. Folgendes verwende ich:

NodeMCU v1.0 (ESP8266)
WS2812B (5m / 150 LEDs)
5V / 40 A Netzteil
Mosquitto als MQTT-Server auf einem Raspberry Pie 3

Ohne aktiviertes AMQTT oder MQTT:
Alles funktioniert wie es soll. AP startet --> WLAN-Daten eintragen --> Dateien hochladen. Nach Neustart alles ok und lässt sich steuern.

Mit aktiviertem AMQTT oder MQTT:
Alles funktioniert erst einmal so, wie es soll. AP startet --> WLAN-Daten + MQTT-Daten eintragen --> Dateien hochladen. Anschließend startet es mit WLAN und verbindet sich zum MQTT-Server. Alles lässt sich steuern (über die Webseite und über FHEM (MQTT-Device).
Es funktioniert aber nur so lange, bis ein Reset durchgeführt wird (oder aber nach entfernen und wiederanschließen der Stromversorgung). Ab diesem Zeitpunkt erhalte ich nur noch exception 28 und eine Endlosschleife mit boot, rst, boot, rst.

Ich habe die aktuellen Versionen der Bibliotheken getestet und ältere (ArduinoJSON geht ja nur auf MASTER-6 und WifiManager ging auf der 0.12.0 nicht - das kommt ein Fehler hinslichtlich WPS).

Sorry, wenn das ggf. einfach nur ein blöder Anfänger-Fehler ist, aber ich komme hier absolut nicht weiter.

Benötigt ihr weitere Infos / Daten?

Vielen Dank im Voraus!
Sascha

**ENG (translated with google-translate --> sorry :-)**

Hello, tobias,
Hello everybody,

I use the following:

NodeMCU v1.0 (ESP8266)
WS2812B (5m / 150 LEDs)
5V / 40A power supply
Mosquitto as MQTT server on a Raspberry Pie 3

Without activated AMQTT or MQTT:
Everything works as it should. AP starts -> enter WLAN data -> upload files. After restart everything ok and can be controlled.

With AMQTT or MQTT enabled:
Everything works the way it should. AP starts -> enter WLAN data + MQTT data -> upload files. Then it starts with WLAN and connects to the MQTT server. Everything can be controlled (via the website and via FHEM (MQTT device).
However, it only works until a reset is performed (or after removing and reconnecting the power supply). From this point on I only get exception 28 and an endless loop with boot, rst, boot, rst.

I've tested the latest versions of the libraries and older ones (ArduinoJSON only works on MASTER-6 and WifiManager did not work on 0.12.0 - that's a bug with WPS).

Sorry, if that's just a stupid beginner mistake, but I'm stuck here.

Do you need more info / data?

Thanks in advance!
Sascha

bmaster001, 2018-12-10:
I think I'm having the same problem.. I'm a newbie to all of this, so any hints in the right direction are very welcome here as well :)

SmilyHM, 2018-12-11:
I'am a newbie, too. I tried to use MQTT with pupsubclient for MQTT and "Async MQTT by @marvinroger" + ESPAsyncTCP for AMQTT. I'm not sure, if it could be a problem with wifimanager (I found some threads wich said it could be probably the problem) - but i am not able to deactivate wifimanager + use a different way for wlan-connection --> cause of "low-skill" :-/

debsahu, 2018-12-11:
[image]
Erase flash completely: All Flash Contents
Flash Size: 4M (3M SPIFFs)
If using AMQTT with Autodiscovery for Home Assistant, Enable IwIP variant: v2 Higher Bandwidth

Ideally use PlatformIO, ditch ArduinoIDE! Included platform.ini is ready to deploy for various platforms including ESP-01.

debsahu, 2018-12-11:
Check [Wiki](https://github.com/toblum/McLighting/wiki/Software-installation) for exact version of libraries that are needed, all of which can be alleviated with PlatformIO. One can manually put in MQTT settings in **defintions.h** 
```
  char mqtt_host[64] = "";
  char mqtt_port[6] = "";
  char mqtt_user[32] = "";
  char mqtt_pass[32] = "";
```

bmaster001, 2018-12-12:
Thanks, I'll try it all later! And I'll also take a look at PlatformIO.. looks promising indeed :-)