toblum/McLighting #286: Home Assistant automatic discovery needs updating

aidbish, 2018-12-11:
In the upcoming version of Home Assistant 0.84. the light .mqtt_json has been merged into light.mqtt.

For people with HA automatic discovery set the lights will not show until this is updated or they manually create the light.

https://rc--home-assistant-docs.netlify.com/blog/2018/12/06/release-84/

https://rc--home-assistant-docs.netlify.com/components/light.mqtt/

debsahu, 2018-12-11:
Thank you for pointing this out, so only thing that has changed is `mqtt_json` should be changed to `mqtt`? I'll add this change to develop version, could you check later today? Keep a lookout for a commit.

debsahu, 2018-12-11:
Could you try the develop branch out and check before we merge this into master. Haven't updated my HA yet to test this.

aidbish, 2018-12-17:
Hi Debsahu
I tried the new version (sorry for the delay) and we still have an error in Home assistant
"platform": "mqtt" is deprecated, replace with "schema":"basic"

i believe there is more to the new mqtt platford  (schema)
https://www.home-assistant.io/components/light.mqtt/#json-schema---configuration

debsahu, 2018-12-17:
I think that is just a warning, are you to control McLighting from HA. I have a patched up HA version with components that I upgraded manually, so can't test.

McLighting >2.1.9 version does indeed send that out unless the sketch also needs extra memory as required by ArduinoJSON v6.7.0. if you want to try, could you add `+ 1500` to 2 locations `DynamicJsonDocument jsonBuffer(JSON_ARRAY_SIZE(strip.getModeCount()) + JSON_OBJECT_SIZE(12)); + 1500` in **request_handlers.h**

aidbish, 2018-12-17:
Hi debashu

tried updating as requested but error is still there. At the moment i agree it just a warning, but if the do deprecate in future the component may stop

debsahu, 2018-12-17:
But can you control McLighting from HA?

aidbish, 2018-12-17:
Hi Debashu

No the light doesnt appear in HA to be able to control it

debsahu, 2018-12-18:
Ok I upgraded mine and broke a few things! Upgrades shouldnt be this painful. So everything is ok on our end, it is just that HA is sending out commands not as json! Trying to figure this thing out. I'm working on it!

rgogada, 2018-12-18:
there is a breaking change in HA 0.84.
All mqtt lights are merged into mqtt (no more mqtt_json)

debsahu, 2018-12-18:
yes we are tracking that change here, I followed what was adviced. added `schema: "json"` sent out to auto-discovery. 

The master branch code sends the following to topic: **homeassistant/light/McLighting01/config**
```
{  
   "name":"McLighting01",
   "platform":"mqtt",
   "schema":"json",
   "state_topic":"home/McLighting01_ha/state/out",
   "command_topic":"home/McLighting01_ha/state/in",
   "on_command_type":"first",
   "brightness":"true",
   "rgb":"true",
   "optimistic":"false",
   "color_temp":"true",
   "effect":"true",
   "effect_list":[  
      "Static",
      "Blink",
      "Breath",
      "Color Wipe",
      "Color Wipe Inverse",
      "Color Wipe Reverse",
      "Color Wipe Reverse Inverse",
      "Color Wipe Random",
      "Random Color",
      "Single Dynamic",
      "Multi Dynamic",
      "Rainbow",
      "Rainbow Cycle",
      "Scan",
      "Dual Scan",
      "Fade",
      "Theater Chase",
      "Theater Chase Rainbow",
      "Running Lights",
      "Twinkle",
      "Twinkle Random",
      "Twinkle Fade",
      "Twinkle Fade Random",
      "Sparkle",
      "Flash Sparkle",
      "Hyper Sparkle",
      "Strobe",
      "Strobe Rainbow",
      "Multi Strobe",
      "Blink Rainbow",
      "Chase White"
   ]
}
```

Any advice would help.

aidbish, 2018-12-18:
> Ok I upgraded mine and broke a few things! Upgrades shouldnt be this painful. So everything is ok on our end, it is just that HA is sending out commands not as json! Trying to figure this thing out. I'm working on it!

Yeah some fairly big breaking changes in last few updates. i think they are trying to get a lot of things in place so that upgrades can be less painful in future and also so people don't have to spend to much time editing files and things can be done from the UI. Yeah mate thanks for looking into this

debsahu, 2018-12-18:
Should be fixed in v1.1.11!

closed 2018-12-18 by debsahu

aidbish, 2018-12-18:
Confirmed working. Thanks Debsahu

aidbish, 2018-12-19:
Hi Debsahu. 
Just found a possible bug.
Whilst the discovery now seems to work and the led is visible in the front end of HA and is controllable. i not the following in the HA error log when i restart the led device.

Error doing job: Exception in callback <bound method MQTT._mqtt_handle_message of <homeassistant.components.mqtt.MQTT object at 0x7fd6d4381518>>

it all seems to work, it just this error when a restart is performed