toblum/McLighting #287: Implement E1.31 sACN into McLighting

debsahu, 2018-12-11:
Idea is to depreciate MeshyMcLighting #147 from experimental branch and instead use E1.31 sACN implementation from https://github.com/forkineye/ESPAsyncE131. Example of such an implementation (non-async) can be found at [WLED](https://github.com/Aircoookie/WLED). We need testers before this is pushed into master branch.

debsahu, 2018-12-12:
Check E131 branch: https://github.com/toblum/McLighting/tree/e131 out!

@toblum Whenever you have time, could you check the modifications to WebUI? All we need to do is send "=e131" via mqtt or websockets. I confirmed that mqtt part works, not sure about WebUI.

We still need people to try > 170 LEDs. I have not tested it beyond my test setup of 16 LEDs.

aidbish, 2018-12-18:
Hi Debsahu.
Have you managed to create video on how to use this new branch? Would be interested to test out

debsahu, 2018-12-18:
I have the video, Adobe Premiere Pro stopped half-way through on Sunday. Forgot to save the workflow, didn't have time to re-edit everything and swamped with work before holidays. I'll try to do this tonight.

DamianFlynn, 2018-12-18:
I have a few NodeMCUs and a new 5 meter coil

toblum, 2018-12-23:
Hi @debsahu,

thank to your great video I was now able to configure the software correctly and everything works as a charm now. I tried a 30 LED strip without any noticeable lag.
The only thing I noticed is that the ESP takes a quite long time after startup (about one minute) to serve webpages for the first time. 

Best regards
Tobias

aidbish, 2018-12-23:
Hi, where can we view the video

toblum, 2018-12-23:
@aidbish You can find it on @debsahu channel: https://www.youtube.com/watch?v=lZ09GlO2_8s

aidbish, 2018-12-23:
ahh yes, already had subscribed to debsahu channel, just not been on youtube for a while

debsahu, 2018-12-24:
> The only thing I noticed is that the ESP takes a quite long time after startup (about one minute) to serve webpages for the first time.
I will see what's causing this.

Todo: modify getModes() instead of modifying web interface.

debsahu, 2019-01-02:
Todo: add Home Assistant mode for E1.31.

toblum, 2019-01-07:
Hi @debsahu,

I just signed up and activated the McLighting repository (https://travis-ci.org/toblum/McLighting/).  I hope this works for you too. I did not find any way to invite others to access the dashboard. Maybe this works automatically for you as you're also an admin.

Regards
Tobias

debsahu, 2019-01-07:
@toblum Yes, it should be automatic as long as there is a .travis.yml file. Once we never this branch, it will trigger a build and you should get emails with build status for every merge.

iansexton, 2019-01-11:
Hi @debsahu & @toblum 

I love this! I discovered McLighting just before Xmas when I needed something to control my lights. Recently I've been trying the E1.31 with Jinx! which has kept me entertained for a while :) I've had a few issues which I'll report when I'm sure it isn't just me being stupid...
Please excuse me plugging my own work, but I have a board which I designed for my students which has an 8 pixel strip. I've found this to be very useful for experimenting with McLighting & Jinx! It's what I used for my Xmas lights. It's at http://www.myiot.co.uk/wemos/

Cheers
Ian

iansexton, 2019-01-15:
I'm happy to report that the issues I had were mine (using Jinx! wrongly). Everything works great now :)
I've had 4 nodes running for several hours synchronised with Jinx! broadcast.
I've noticed that new nodes can struggle to connect to my access point when a 'show' is running but I guess this is down to my access point. Superb work guys.

Cheers
Ian

debsahu, 2019-01-15:
When E1.31 packets arrive, first preference is for the packets and one safe webserver loop packet. That's why your are seeing the show down. There is no solution for it, unless webserver is Async as well.

iansexton, 2019-01-15:
> I have tested this branch with Xlights - and its working like magic -
> This is going to be very interesting

It's excellent!

My only reservation is that the array size is hard coded in McLighting. 
This needs a little thought.

debsahu, 2019-01-15:
> It's excellent! My only reservation is that the array size is hard coded in McLighting. This needs a little thought.

We can use pointers for strip and need significant changes to WebUI for getting strip length and maybe RGB order and such. It is different topic, as of now, just focusing on E1.31 in this PR.

closed 2019-01-23 by debsahu