toblum/McLighting #290: Repo cleanup

M4lik, 2018-12-12:
Hey there. It's me again ^^
I just wanted to suggest some small changes to the repo for two reasons: Make it a little bit more appealing to the eye and also make it easier for someone who just found out about this project, to get around.
This would include:

- Clean up the [`README.md`](https://github.com/toblum/McLighting/blob/7a44a76aca74d8501f6018c1d9dfc28a43ae45ee/README.md)
  - Move all news entries to their own file
  - Rewriting some paragraphs
- Starting to use releases
  - Since github created a new watch-mode called "Releases only" where you get a mail everytime a new release is being published, I would recommend using this feature ^^
  - Furthermore since it's possible to

toblum, 2018-12-16:
@M4lik Thank you for doing the frontpage cleanup.
I have to look into releases, didn't you the feature in GitHub before.
Polishing the wiki is something I have also on my list but didn't find time yet. If you got some enhancements feel free to do them. Are you able to edit the wiki? I'm not sure how this is set.

toblum, 2018-12-18:
Hi @M4lik,

I invited you as a collaborator, so you should be able to access and edit the wiki now. That's much easier. Regarding releases: This is easy. I didn't had that in mind as I'm using GitHUb just for my private repos.

Will look into the other things soon.

Regards
Tobias

toblum, 2018-12-23:
Having motivated people helping is great! This keeps projects like this running.

Sure I will keep the issue open.

Regards
Tobias

aidbish, 2018-12-27:
This project is probably the best LED one on Github. especially even more so with the enhancements @debsahu is bringing with the E1.31 support. this will make Mclighting more appealing to a lot more people.

Keep up the great work @toblum & @debsahu

M4lik, 2019-01-07:
Just a quick update: I've now refurbished the HTTP API doc.
It came to my attention that often the color and speed are defined twice like [here](https://github.com/toblum/McLighting/wiki/HTTP-REST-API#get-lightmode-http10--)

Is there e reason behind that? Or is it just convenience? (Yes, I've seen that it's here an integer and there a hex value)
If it's the latter one I would _strongly_ recommend to change that to only one, because:

1. It's very bad practice to have a required parameter that excludes another required parameter
2. Seen from an API point of view, it's basically undefined behavior if someone is setting both params.

The same applies to the brightness parameter (`c` and `p`).

And one more thing: It's also not good practice to have the same effect emerge from setting two "different" settings. We do this with [`set_speed`](https://github.com/toblum/McLighting/wiki/HTTP-REST-API#get-set_speed-http10) and [`<lightmode>`](https://github.com/toblum/McLighting/wiki/HTTP-REST-API#get-lightmode-http10--) where it's one time speed (higher = faster) and delay (higher = slower)

I know that `<lightmode>` is deprecated, but still...

## Suggestion

My suggestion here is simple but, not wanting to brag, quite good. I would suggest adding a new sub-path like `/api/[endpoint]`. This will allow us to modify the API without breaking anyones existing setup.
To carry this idea even further into the future it may be a good idea to also add a version sub-path like `/api/v2/[endpoint]`.

As I said this is sugst a **suggestion** so I don't really mind if you say this is to far off the track.

Anyway I will also do this kind of post again for Websocket and MQTT if I think it is necessary.

PS:
Yeah I said that a try to start the clean up of the wiki before Christmas. Sorry for that :sweat_smile:

debsahu, 2019-01-07:
OMG, loving the Wiki clean up @M4lik. Can we add a FAQ? Usually people ask about changing color order, setting static IP, etc.

M4lik, 2019-01-07:
Thanks :)
Sure thing. I will add a new page in a few minutes. You can then either provide me some Q+A or you just copy my entry and edit the text :)

M4lik, 2019-01-07:
@debsahu I've added the Page. Just modify it :)

https://github.com/toblum/McLighting/wiki/%5BFAQ%5D-You-ask,-we-answer!

toblum, 2019-01-07:
Hi @M4lik,

the REST-API docs look great. Thank you. Very neat layout. No stress, just do it as you find time.

The API parameters are mostly there for historical reasons. To have the ability to set the colors via hex or RGB, the brightness as percentage or byte value. I have no idea how many people are using these so I have kept them so far. Having an API version in path is great but in the beginning there were so many changes we would have now version 350. :-) Now, as the API is more stable we could think about it for future changes, but I would recommend to keep the existing endpoints as long as possible to not frustrate the users.

The FAQ is a great idea. It can grow over time. Maybe we should link it on the readme, because many people are missing the wiki documentation.

Feel free to add the releases. I think we should do it everytime we do a merge into master that contains essential changes.

Best regards
Tobias

closed 2019-01-24 by debsahu