toblum/McLighting #292: Cycle through modes?

mr-bridger, 2018-12-12:
Hi, just managed to set up McLighting on a D1 Mini :-)  Is there a way to get it to cycle  through all modes? Maybe spend 1-2 min on each? 

(Also, can't for the life of me set the IP Static if I uncomment the line #define WIFI_SET_MANUAL_IP in definitions.h, It just goes back to AP mode. is there a step by step anywhere?)

debsahu, 2018-12-13:
### Auto Mode

Read the wiki: https://github.com/toblum/McLighting/wiki/Autocycling

-------------
### Static IP
Uncomment this:
```
#define WIFIMGR_SET_MANUAL_IP
```
_ip is the IP you want to set, _gw is gateway (usually the router IP), _sn is subnet (usually no change is required)
```
#ifdef WIFIMGR_SET_MANUAL_IP
  uint8_t _ip[4] = {192,168,0,128};
  uint8_t _gw[4] = {192,168,0,1};
  uint8_t _sn[4] = {255,255,255,0};
#endif
```

mr-bridger, 2018-12-13:
Thanks, I missed that. Is that what is toggled with the 'Auto' switch on the colour wheel page?

M4lik, 2018-12-13:
> Thanks, I missed that. Is that what is toggled with the 'Auto' switch on the colour wheel page?

Correct

closed 2018-12-13 by debsahu