toblum/McLighting #294: Constant Exceptions

Spankythemusical, 2018-12-16:
Maybe I'm doing this completely wrong, but I've tried everything.  I loaded the sketch and additional items to my ESP8266.   Everything uploads fine.   Once the upload is done, however, all I experience is constant resets once the WiFi connection is finalized.  I've commented random items out, but nothing has helped, just constant resets.

Here is the serial output:

`*WM: Adding parameter
*WM: host
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: user
*WM: Adding parameter
*WM: pass
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: (IP unset)
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.86.201
SaveConfig is False!
WiFiManager config FS Save failure!
connected...yeey :)
Arduino OTA activated.

MQTT active: 192.168.**.**:1883
Open http://192.168.**.**/ to open McLighting.
Use http://McLighting01.local/ when you have Bonjour installed.
New users: Open http://192.168.**.**/upload to upload the webpages first.

Read cfg: {"mode":1,"strip_mode":45,"brightness":9,"speed":196,"red":255,"green":177,"blue":109} Success!
*WM: freeing allocated params!
MQTT disconnected, reconnecting!
Attempting MQTT connection 1 / 4 ...
MQTT connected!
MQTT topic in: McLighting01/in
MQTT topic out: McLighting01/out

Exception (28):
epc1=0x40214421 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont 
sp: 3fff3190 end: 3fff3450 offset: 01a0
3fff3330:  00000000 00000002 00000000 fffffffe  
3fff3340:  ffffffff 3fffc6fc 00000001 000000e5  
3fff3350:  00000000 00000000 3fff0914 402259a0  
3fff3360:  00000000 00000000 00000000 00000000  
3fff3370:  401050e2 001baf1d 40418937 00000000  
3fff3380:  4010519e 3ffea9d4 001baf1d 00000000  
3fff3390:  40105370 3fff0cc0 3fff0cf8 40221f52  
3fff33a0:  3ffefed0 3fff1e48 3fff4c24 4022b274  
3fff33b0:  000000b8 00000000 00001388 40100808  
3fff33c0:  00000000 3fff7434 00000018 3fff1fbc  
3fff33d0:  3fffdad0 3fff0cc0 3fff7184 40221fa2  
3fff33e0:  4022cf20 00000000 00001388 40228f98  
3fff33f0:  3fffdad0 00000000 3fff0cc0 402227f4  
3fff3400:  3fffdad0 00000000 3fff1f88 40210625  
3fff3410:  00000000 00000000 00000000 feefeffe  
3fff3420:  00000000 00000000 00000001 3fff1fbc  
3fff3430:  3fffdad0 00000000 3fff1f88 40229d38  
3fff3440:  feefeffe feefeffe 3ffe8824 40100a65  
<<<stack<<<
ºP9FS File: /index2.htm, size: 668.69KB
FS File: /stripstate.json, size: 86B
FS File: /edit.htm.gz, size: 4.02KB
FS File: /favicon.ico, size: 1.12KB
FS File: /graphs.js.gz, size: 1.92KB
FS File: /index.htm, size: 13.95KB
FS Usage: 714095/957314 bytes

Coudnt find config.json
WiFiManager config FS Read failure!`

The Exception Decoder shows:

`Decoding 14 results
0x402259a0: WS2812FX::fade_out() at C:\Users\Jerrit\Documents\Arduino\libraries\WS2812FX-master\src/WS2812FX.cpp line 916
0x401050e2: wdt_feed at ?? line ?
0x4010519e: ets_timer_setfn at ?? line ?
0x40105370: ets_timer_arm_new at ?? line ?
0x40221f52: WebSocketsServer::newClient(WiFiClient*) at C:\Users\Jerrit\Documents\Arduino\libraries\arduinoWebSockets-master\src/WebSocketsServer.cpp line 494
0x4022b274: WiFiServer::available(unsigned char*) at C:\Users\Jerrit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta1\libraries\ESP8266WiFi\src/WiFiServer.cpp line 116 (discriminator 2)
0x40100808: malloc at C:\Users\Jerrit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta1\cores\esp8266\umm_malloc/umm_malloc.c line 1672
0x40221fa2: WebSocketsServer::handleNewClients() at C:\Users\Jerrit\Documents\Arduino\libraries\arduinoWebSockets-master\src/WebSocketsServer.cpp line 642
0x4022cf20: DNSServer::setErrorReplyCode(DNSReplyCode const&) at ?? line ?
0x40228f98: String::invalidate() at C:\Users\Jerrit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta1\cores\esp8266/WString.cpp line 141
0x402227f4: WebSocketsServer::loop() at C:\Users\Jerrit\Documents\Arduino\libraries\arduinoWebSockets-master\src/WebSocketsServer.cpp line 133
0x40210625: loop at C:\Users\Jerrit\Documents\Arduino\McLighting/McLighting.ino line 922
0x40229d38: loop_wrapper() at C:\Users\Jerrit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta1\cores\esp8266/core_esp8266_main.cpp line 125
0x40100a65: cont_wrapper at C:\Users\Jerrit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta1\cores\esp8266/cont.S line 81`

And here's the top section of my definitions:

`//#define USE_WS2812FX_DMA      // Uses PIN is ignored & set to RX/GPIO3  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX
//#define USE_WS2812FX_UART     // Uses PIN is ignored & set to D4/GPIO2  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX

// Neopixel
#define PIN D6           // PIN (14 / D5) where neopixel / WS2811 strip is attached
#define NUMLEDS 230       // Number of leds in the strip
//#define BUILTIN_LED 2    // ESP-12F has the built in LED on GPIO2, see https://github.com/esp8266/Arduino/issues/2192
//#define BUTTON 4         // Input pin (4 / D2) for switching the LED strip on / off, connect this PIN to ground to trigger button.

const char HOSTNAME[] = "McLighting01";   // Friedly hostname

#define HTTP_OTA             // If defined, enable ESP8266HTTPUpdateServer OTA code.
#define ENABLE_OTA         // If defined, enable Arduino OTA code.
//define ENABLE_AMQTT         // If defined, enable Async MQTT code, see: https://github.com/marvinroger/async-mqtt-client
#define ENABLE_MQTT        // If defined, enable MQTT client code, see: https://github.com/toblum/McLighting/wiki/MQTT-API
#define ENABLE_HOMEASSISTANT // If defined, enable Homeassistant integration, ENABLE_MQTT or ENABLE_AMQTT must be active
//#define ENABLE_BUTTON        // If defined, enable button handling code, see: https://github.com/toblum/McLighting/wiki/Button-control
//#define MQTT_HOME_ASSISTANT_SUPPORT // If defined, use AMQTT and select Tools -> IwIP Variant -> Higher Bandwidth
#define ENABLE_LEGACY_ANIMATIONS

//#define WIFIMGR_PORTAL_TIMEOUT 180
//#define WIFIMGR_SET_MANUAL_IP

#ifdef WIFIMGR_SET_MANUAL_IP
  uint8_t _ip[4] = {192,168,0,128};
  uint8_t _gw[4] = {192,168,0,1};
  uint8_t _sn[4] = {255,255,255,0};
#endif

#ifdef MQTT_HOME_ASSISTANT_SUPPORT
  #define MQTT_HOME_ASSISTANT_0_84_SUPPORT // Comment if using HA version < 0.84 
#endif

#if defined(USE_WS2812FX_DMA) and defined(USE_WS2812FX_UART)
#error "Cant have both DMA and UART method."
#endif
#if defined(ENABLE_MQTT) and defined(ENABLE_AMQTT)
#error "Cant have both PubSubClient and AsyncMQTT enabled. Choose either one."
#endif
#if ( (defined(ENABLE_HOMEASSISTANT) and !defined(ENABLE_MQTT)) and (defined(ENABLE_HOMEASSISTANT) and !defined(ENABLE_AMQTT)) )
#error "To use HA, you have to either enable PubCubClient or AsyncMQTT"
#endif
#if ( !defined(ENABLE_HOMEASSISTANT) and defined(MQTT_HOME_ASSISTANT_SUPPORT) )
#error "To use HA support, you have to either enable Homeassistant component"
#endif

// parameters for automatically cycling favorite patterns
uint32_t autoParams[][4] = { // color, speed, mode, duration (seconds)
  {0xff0000, 200,  1,  5.0}, // blink red for 5 seconds
  {0x00ff00, 200,  3, 10.0}, // wipe green for 10 seconds
  {0x0000ff, 200, 11,  5.0}, // dual scan blue for 5 seconds
  {0x0000ff, 200, 42, 15.0}  // fireworks for 15 seconds
};

#if defined(ENABLE_MQTT) or defined(ENABLE_AMQTT)
  #ifdef ENABLE_MQTT
    #define MQTT_MAX_PACKET_SIZE 2048
    #define MQTT_MAX_RECONNECT_TRIES 4

    int mqtt_reconnect_retries = 0;
    char mqtt_intopic[strlen(HOSTNAME) + 4 + 5];      // Topic in will be: <HOSTNAME>/in
    char mqtt_outtopic[strlen(HOSTNAME) + 5 + 5];     // Topic out will be: <HOSTNAME>/out
    uint8_t qossub = 0; // PubSubClient can sub qos 0 or 1
  #endif

  #ifdef ENABLE_AMQTT
    String mqtt_intopic = String(HOSTNAME) + "/in";
    String mqtt_outtopic = String(HOSTNAME) + "/out";
    uint8_t qossub = 0; // AMQTT can sub qos 0 or 1 or 2
    uint8_t qospub = 0; // AMQTT can pub qos 0 or 1 or 2
  #endif

  #ifdef ENABLE_HOMEASSISTANT
    String mqtt_ha = "home/McLighting01_ha/";
    String mqtt_ha_state_in = "home/McLighting01_ha/state/in";
    String mqtt_ha_state_out = "home/McLighting01_ha/state/out";

    const char* on_cmd = "ON";
    const char* off_cmd = "OFF";
    bool stateOn = false;
    bool animation_on = false;
    bool new_ha_mqtt_msg = false;
    uint16_t color_temp = 327; // min is 154 and max is 500
  #endif

  //#define ENABLE_MQTT_HOSTNAME_CHIPID          // Uncomment/comment to add ESPChipID to end of MQTT hostname
  #ifdef ENABLE_MQTT_HOSTNAME_CHIPID
    char mqtt_clientid[64];
  #else
    const char* mqtt_clientid = HOSTNAME;
  #endif

  char mqtt_host[64] = "192.168.**.**";
  char mqtt_port[6] = "1883";
  char mqtt_user[32] = "******";
  char mqtt_pass[32] = "********";
#endif`

Any ideas???

debsahu, 2018-12-16:
What is your sketch version? check **version.h**

debsahu, 2018-12-16:
Try changing `#define MQTT_MAX_PACKET_SIZE 2048` to `#define MQTT_MAX_PACKET_SIZE 512` and see if it fixes things?

Spankythemusical, 2018-12-16:
The sketch version is  2.1.9.  Changed the buffer size to no avail.  After approximately 3 seconds, this is what the serial monitor showed...

`File: /index2.htm, size: 668.69KB
FS File: /stripstate.json, size: 86B
FS File: /edit.htm.gz, size: 4.02KB
FS File: /favicon.ico, size: 1.12KB
FS File: /graphs.js.gz, size: 1.92KB
FS File: /index.htm, size: 13.95KB
FS Usage: 714095/957314 bytes

Coudnt find config.json
WiFiManager config FS Read failure!
*WM: Adding parameter
*WM: host
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: user
*WM: Adding parameter
*WM: pass
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: (IP unset)
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.86.201
SaveConfig is False!
WiFiManager config FS Save failure!
connected...yeey :)
Arduino OTA activated.

MQTT active: 192.168.86.78:1883
Open http://192.168.86.201/ to open McLighting.
Use http://McLighting01.local/ when you have Bonjour installed.
New users: Open http://192.168.86.201/upload to upload the webpages first.

Read cfg: {"mode":1,"strip_mode":45,"brightness":9,"speed":196,"red":255,"green":177,"blue":109} Success!
*WM: freeing allocated params!
MQTT disconnected, reconnecting!
Attempting MQTT connection 1 / 4 ...
MQTT connected!
MQTT topic in: McLighting01/in
MQTT topic out: McLighting01/out
WS: [0] Disconnected!

Exception (28):
epc1=0x40214421 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont 
sp: 3fff2b90 end: 3fff2e50 offset: 01a0
3fff2d30:  ffffffff 00000002 00000001 3fff170c  
3fff2d40:  00000000 3fffdad0 3fff19bc 000000e5  
3fff2d50:  00000000 00000000 3fff0914 402259a0  
3fff2d60:  00000000 00000000 00000000 00000000  
3fff2d70:  401050e2 00650557 f47ae147 00000000  
3fff2d80:  4010519e 3ffefea8 00650557 00000000  
3fff2d90:  40105370 3fff0cc0 3fff0cf8 40221f52  
3fff2da0:  3ffefed0 3fff1848 3fff4624 4022b274  
3fff2db0:  000000b8 00000000 00001388 40100808  
3fff2dc0:  00000000 3fff359c 00000018 3fff19bc  
3fff2dd0:  3fffdad0 3fff0cc0 3fff6ae4 40221fa2  
3fff2de0:  4022cf20 00000000 00001388 40228f98  
3fff2df0:  3fffdad0 00000000 3fff0cc0 402227f4  
3fff2e00:  3fffdad0 00000000 3fff1988 40210625  
3fff2e10:  00000000 00000000 00000000 feefeffe  
3fff2e20:  00000000 00000000 00000001 3fff19bc  
3fff2e30:  3fffdad0 00000000 3fff1988 40229d38  
3fff2e40:  feefeffe feefeffe 3ffe8824 40100a65  
<<<stack<<<
`
Exception Decoder shows:

`0x402259a0: WS2812FX::fade_out() at C:\Users\Jerrit\Documents\Arduino\libraries\WS2812FX-master\src/WS2812FX.cpp line 916
0x401050e2: wdt_feed at ?? line ?
0x4010519e: ets_timer_setfn at ?? line ?
0x40105370: ets_timer_arm_new at ?? line ?
0x40221f52: WebSocketsServer::newClient(WiFiClient*) at C:\Users\Jerrit\Documents\Arduino\libraries\arduinoWebSockets-master\src/WebSocketsServer.cpp line 494
0x4022b274: WiFiServer::available(unsigned char*) at C:\Users\Jerrit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta1\libraries\ESP8266WiFi\src/WiFiServer.cpp line 116 (discriminator 2)
0x40100808: malloc at C:\Users\Jerrit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta1\cores\esp8266\umm_malloc/umm_malloc.c line 1672
0x40221fa2: WebSocketsServer::handleNewClients() at C:\Users\Jerrit\Documents\Arduino\libraries\arduinoWebSockets-master\src/WebSocketsServer.cpp line 642
0x4022cf20: DNSServer::setErrorReplyCode(DNSReplyCode const&) at ?? line ?
0x40228f98: String::invalidate() at C:\Users\Jerrit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta1\cores\esp8266/WString.cpp line 141
0x402227f4: WebSocketsServer::loop() at C:\Users\Jerrit\Documents\Arduino\libraries\arduinoWebSockets-master\src/WebSocketsServer.cpp line 133
0x40210625: loop at C:\Users\Jerrit\Documents\Arduino\McLighting/McLighting.ino line 922
0x40229d38: loop_wrapper() at C:\Users\Jerrit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta1\cores\esp8266/core_esp8266_main.cpp line 125
0x40100a65: cont_wrapper at C:\Users\Jerrit\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta1\cores\esp8266/cont.S line 81`

Spankythemusical, 2018-12-18:
After trying everything else, I just swapped to a new NodeMCU.  Had one exception restart, but it's been solid since then.  I'll just use the other one for something else.

closed 2018-12-18 by Spankythemusical