toblum/McLighting #296: Wemos restart when using webinterface

gent99, 2018-12-18:
HI,

one of our wemos D1 mini shows this stack trace when we try to use the web interface (V2):

Could it be that the wemos is damaged, somehow? More than 30 others are working fine so far.

Kind regards, Holger

Panic /Users/menzel/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WiFi/src/include/DataSource.h:93 const uint8_t* BufferedStreamDataSource<TStream>::get_buffer(size_t) [with TStream = Stream; uint8_t = unsigned char; size_t = unsigned int]: Assertion 'cb == stream_rem' failed.
ctx: cont 
sp: 3ffffbd0 end: 3fffffd0 offset: 01b0
>>>stack>>>
3ffffd80:  3fff53b4 00000100 3fff22f0 402031c5  
3ffffd90:  3fff3044 00000000 3fff3824 40212745  
3ffffda0:  3fff17e0 00000603 00000603 00000100  
3ffffdb0:  00000100 00000000 3fff4b04 40223efd  
3ffffdc0:  00000100 4022387c 00000100 40225c84  
3ffffdd0:  3fff3044 4021238c 00000100 402079b4  
3ffffde0:  00000002 00000000 3fff4b04 402123ca  
3ffffdf0:  00000000 4bc6a7f0 00001610 00000000  
3ffffe00:  00000100 00000af1 3fff3824 40212651  
3ffffe10:  00000000 3fff11a0 401066c2 c2d0e560  
3ffffe20:  00000000 4022384c 0000001c 0003da5a  
3ffffe30:  3fff3824 3ffffe70 3fff3824 4021270c  
3ffffe40:  00000001 3ffffe70 00000000 40212a90  
3ffffe50:  00000000 00000000 3fff11a0 3ffe91a5  
3ffffe60:  00000001 0003da5a 3fff11a0 4020a626  
3ffffe70:  402263a8 00000000 000003e8 3fff169c  
3ffffe80:  3fff3c24 3fff385c 00000000 00000000  
3ffffe90:  00000000 00000000 00000000 00000000  
3ffffea0:  3fff4aec 0000000f 0000000d 3fff3874  
3ffffeb0:  0000000f 00000009 3ffffee0 40222e80  
3ffffec0:  00000000 00000000 3fff11a0 3fff169c  
3ffffed0:  00000000 3fff11e4 3fff11a0 4020a6c4  
3ffffee0:  3fff380c 0000000f 0000000a 00000000  
3ffffef0:  00000000 00000000 3fff11a0 40215046  
3fffff00:  00000000 00000000 00000000 4021516e  
3fffff10:  00000000 00000000 00000000 00000000  
3fffff20:  00000000 4bc6a7f0 00001567 3fff169c  
3fffff30:  3fff11e4 00000001 4bc6a7f0 00000000  
3fffff40:  00000000 00000000 402138b0 0000000f  
3fffff50:  00000000 00000000 4bc6a7f0 3fff169c  
3fffff60:  00000001 3fff11c8 3fff11a0 40215358  
3fffff70:  00000000 00000000 00001388 3fff169c  
3fffff80:  3fffdad0 00000000 3fff1696 4020a329  
3fffff90:  3fffdad0 00000000 3fff1696 4020f85b  
3fffffa0:  00000000 00000000 00000001 3fff169c  
3fffffb0:  3fffdad0 00000000 3fff1696 40223a14  
3fffffc0:  feefeffe feefeffe 3ffe881c 40100739  
<<<stack<<<
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
FS File: /stripstate.json, size: 83B
FS File: /index.htm, size: 246.59KB
FS File: /uistate.json, size: 75B
FS Usage: 872476/2949250 bytes

debsahu, 2018-12-18:
Re-upload firmware?

gent99, 2018-12-18:
We uploaded all: sketch and files and now it‘s working again. So please close or delete this issue. Many thanks, Holger

closed 2018-12-18 by debsahu