toblum/McLighting #298: MQTT-Status / Module Status

ozett, 2018-12-21:
i tried your firmware: its great.

as i newbie, i am missing some information about the running status.
at least at tasmota/sonoff there is a textpage. 
Maybe you want to code this a a new feature?

[image]

i was trying to check the mqtt status, and saw different topis on console.
is this by design? or a bug?

[image]

toblum, 2018-12-23:
Hi @ozett,

having some more informations in the UI is definitely on the list.
There is at least an API endpoint that gives some configuration and runtime informations: http://<IP_ADRESS>/esp_status

And yes, the MQTT topics are there by design. Some of them are there if you enables homeassistant integration: https://github.com/toblum/McLighting/wiki/Homeassistant-integration

Regards
Tobias

closed 2018-12-23 by toblum

ozett, 2018-12-24:
just a sidenote: 

the homeassistant-mqtt-topis should not be there,
because i did _not_ enable it...

(think thats the purpose of enable/disable)

debsahu, 2018-12-24:
What's your output of http://<IP_ADRESS>/esp_status

ozett, 2018-12-25:
[image]