toblum/McLighting #299: Need help troubleshooting modes

cwesterfield, 2018-12-23:
I just got a line of 300 working and some modes sort of devolve into the whole string just blinking white really fast. 

Modes that definitely work are Merry Christmas and Chase Blackout Rainbow. The Twinkle ones does seems to work. I can post a video if it would help. 

The Chase seemed like a good test, because it gets all the way to 300 and then starts over, so it seems like my esp can control each one. Am I maybe overloading the memory when trying to dynamically effect 300 lights at once?

- https://www.amazon.com/gp/product/B01M0KLECZ
- https://www.amazon.com/gp/product/B073D4DJDC
- https://www.amazon.com/gp/product/B07FVR781Z
- https://www.amazon.com/gp/product/B010O1G1ES

cwesterfield, 2018-12-23:
I just did the steps in the software setup, while commenting out the mqtt stuff. 

I also changed the pin for data to how it is on your hardware schematic (D1). 

I'll check the DMA and UART Stuff.

debsahu, 2018-12-24:
The issue is maybe that there is no common ground with ESP8266 or noisy power supply. First one can be fixed by connecting the ground of strip to ground of ESP8266. Second can be fixed with using multiple capacitors in parallel to 5V supply. @cwesterfield were you able to perform [Adafruit NoePixel's strandtest](https://github.com/adafruit/Adafruit_NeoPixel/tree/master/examples/strandtest) with no issues?

cwesterfield, 2018-12-24:
Using DMA it acts quite a bit better. I'm not sure if UART would be any change. 

I'm not sure how to do the strandtest, so I'll look into it. 

Is there a fritz outlining the grounds you are mentioning?

debsahu, 2018-12-24:
https://github.com/toblum/McLighting/wiki/Hardware

debsahu, 2018-12-24:
Add a 1000uF, 10uF electrolytic and 22nF ceramic (you can use any values, vary the kind of caps and values) caps between 5V and ground. Should get rid of most ripples, assuming you have one. Try to do the stand test first.

cwesterfield, 2019-01-29:
Bad luck on the new power supply. DMA fixes it thought.

closed 2019-01-29 by cwesterfield