toblum/McLighting #300: Can't add second McLightning device in Home Assistant

hristo-atanasov, 2018-12-27:
`  light:
- platform: mqtt
    schema: json
    host: 192.168.0.16
    name: Left Led Strip
    state_topic: "home/LeftLedStrip01_ha/state/out"
    command_topic: "home/LeftLedStrip01_ha/state/in"
    on_command_type: "first"
    brightness: true
    rgb: true
    optimistic: false
    color_temp: true
    effect: true
    effect_list:
        - Static
        - Blink
        - Breath
        - Color Wipe
        - Color Wipe Inverse
        - Color Wipe Reverse
        - Color Wipe Reverse Inverse
        - Color Wipe Random
        - Random Color
        - Single Dynamic
        - Multi Dynamic
        - Rainbow
        - Rainbow Cycle
        - Scan
        - Dual Scan
        - Fade
        - Theater Chase
        - Theater Chase Rainbow
        - Running Lights
        - Twinkle
        - Twinkle Random
        - Twinkle Fade
        - Twinkle Fade Random
        - Sparkle
        - Flash Sparkle
        - Hyper Sparkle
        - Strobe
        - Strobe Rainbow
        - Multi Strobe
        - Blink Rainbow
        - Chase White
        - Chase Color
        - Chase Random
        - Chase Rainbow
        - Chase Flash
        - Chase Flash Random
        - Chase Rainbow White
        - Chase Blackout
        - Chase Blackout Rainbow
        - Color Sweep Random
        - Running Color
        - Running Red Blue
        - Running Random
        - Larson Scanner
        - Comet
        - Fireworks
        - Fireworks Random
        - Merry Christmas
        - Fire Flicker
        - Fire Flicker (soft)
        - Fire Flicker (intense)
        - Circus Combustus
        - Halloween
        - Bicolor Chase
        - Tricolor Chase
        - ICU
        - Custom 0
        - Custom 1
        - Custom 2
        - Custom 3
  - platform: mqtt
    schema: json
    host: 192.168.0.17
    name: Right Led Strip
    state_topic: "home/RightLedStrip01_ha/state/out"
    command_topic: "home/RightLedStrip01_ha/state/in"
    on_command_type: "first"
    brightness: true
    rgb: true
    optimistic: false
    color_temp: true
    effect: true
    effect_list:
        - Static
        - Blink
        - Breath
        - Color Wipe
        - Color Wipe Inverse
        - Color Wipe Reverse
        - Color Wipe Reverse Inverse
        - Color Wipe Random
        - Random Color
        - Single Dynamic
        - Multi Dynamic
        - Rainbow
        - Rainbow Cycle
        - Scan
        - Dual Scan
        - Fade
        - Theater Chase
        - Theater Chase Rainbow
        - Running Lights
        - Twinkle
        - Twinkle Random
        - Twinkle Fade
        - Twinkle Fade Random
        - Sparkle
        - Flash Sparkle
        - Hyper Sparkle
        - Strobe
        - Strobe Rainbow
        - Multi Strobe
        - Blink Rainbow
        - Chase White
        - Chase Color
        - Chase Random
        - Chase Rainbow
        - Chase Flash
        - Chase Flash Random
        - Chase Rainbow White
        - Chase Blackout
        - Chase Blackout Rainbow
        - Color Sweep Random
        - Running Color
        - Running Red Blue
        - Running Random
        - Larson Scanner
        - Comet
        - Fireworks
        - Fireworks Random
        - Merry Christmas
        - Fire Flicker
        - Fire Flicker (soft)
        - Fire Flicker (intense)
        - Circus Combustus
        - Halloween
        - Bicolor Chase
        - Tricolor Chase
        - ICU
        - Custom 0
        - Custom 1
        - Custom 2
        - Custom 3`

debsahu, 2018-12-27:
Both can't have the same name, change the hostname in definitions.h

hristo-atanasov, 2018-12-27:
They don't seem to have same names, do they? :\

hristo-atanasov, 2018-12-27:
The first is 'LeftLedStrip0' and the second is 'RightLedStrip0'

aidbish, 2018-12-27:
Install a mqtt app like MQTT.fx, 
Connect to your MQTT broker and subscribe to home/#  you should be able to see if the second light is connecting to your broker when the device is powered on

hristo-atanasov, 2018-12-27:
It was my mistake. It is working now. I didn't realize there was a newer version with 0.84.x support.

debsahu, 2018-12-27:
Reopen if you still got issues

closed 2018-12-27 by debsahu