toblum/McLighting #302: Request: Compatibility with ESP-01S

redfirebrooks, 2018-12-31:
Hello,
great piece of code and works perfectly.
I'd like to miniaturise and use an ESP-01S instead of a WEMOS D1 mini. Could it be possible ?

debsahu, 2019-01-02:
You will need at least 1M Flash chip, you can upgrade FLASH to 4M by following instructions here: https://www.youtube.com/watch?v=xyc1gCjguRU . Why ESP01S? They are not cheaper than ESP12E, ESP12F. OTA will not be possible with 1M flash! My suggestion is to use ESP8285 ESP-M3 which have 1M Flash and are very compact, but same OTA limitations.

This project is compatible with 1M Flash as the compilation succeeds in the platformio for [env:esp01_1m] ([platformio.ini](https://github.com/toblum/McLighting/blob/master/platformio.ini))

debsahu, 2019-02-05:
closing due to inactivity, solution is given in previous reply. Please reopen if you still face issues.

closed 2019-02-05 by debsahu