toblum/McLighting #303: Green and Blue channels backwards

patrol420, 2019-01-04:
ws2812b light strip, red is fine but green and blue is backwards. im assuming its because of slight variations in strips? can i swap them in the firmware so the colours come out properly?

debsahu, 2019-01-04:
In **McLighting.ino**, change `WS2812FX strip = WS2812FX(NUMLEDS, PIN, NEO_GRB + NEO_KHZ800);` to `WS2812FX strip = WS2812FX(NUMLEDS, PIN, NEO_RGB + NEO_KHZ800);`

pflaugh, 2019-01-05:
This really helped me! Thanks @debsahu.

closed 2019-01-05 by debsahu

jersam, 2019-04-26:
```
// ***************************************************************************
// Load libraries / Instanciate WS2812FX library
// ***************************************************************************
// https://github.com/kitesurfer1404/WS2812FX
#include <WS2812FX.h>
// WS2812FX strip = WS2812FX(NUMLEDS, PIN, NEO_BRG + NEO_KHZ800);
WS2812FX* strip;
```

when I uncomment the fix line I get the following error...

```
exit status 1
'PIN' was not declared in this scope
```

I am using #define USE_WS2812FX_DMA in definitions.h