toblum/McLighting #308: Please help with one button cycling through modes

KayCox, 2019-01-07:
Please help me coding pros. ;)

Is it possible to toggle trough some defined modes with the same button? Need more than 3 modes as usual in regular button handling script (my one below) and I want every short press going to the next defined mode. 
Anybody there who could give me a coding example on how to do that?

Thanks for your help.

// Button handling
#ifdef ENABLE_BUTTON
#define BTN_MODE_SHORT "STA| 1| 0|245|196|0|255|0" // Static green
#define BTN_MODE_MEDIUM "STA| 1| 0|245|196|255|0|0" // Static red
#define BTN_MODE_LONG "STA| 1| 46|253|196|255|102| 0" // Fireworks random

debsahu, 2019-02-01:
Closing due to inactivity. The question is answered.

closed 2019-02-01 by debsahu

odis68, 2019-12-28:
Please help step by step what and what to replace to activate the automatic program by a long press on the button.